tomvdb/open_tuner #3: Open Tuner

g6hfs, 2023-03-18:
Hi Tom,

What about adding the signal strenth into the satallie, when mouse poiner hovers over signal bar.

Brian (G6HFS)

tomvdb, 2023-03-19:
Can definitely look into that :)

iu4gav, 2023-03-24:
It seems to me a good decryption software alternative to minitioune, except that:

 1) below 66 ksr it does not decode, by choice of whoever made the software! (in the source file the 35 ksr is set wrong and the ksr below 35 are not managed at all) 

2) it would be good to have the threshold level to check if you are transmitting enough to be decoded or if you are low on power! 

3) in the Mer you need decimals, which would be fundamental if you decide to use open-tuner to point the dish!

 73
 IU4GAV

dl1rf, 2024-11-22:
> Hi Tom,
> 
> What about adding the signal strenth into the satallite, when mouse pointer hovers over a signal bar.
> 
> Brian (G6HFS)

I have added the signal strength feature.
Signal strength is in dB against beacon (dbB).
feature is included in my latest pull request.

Martin

g6hfs, 2024-11-23:
Hi,Can not find download link.73Brian (G6HFS)

dl1rf, 2024-11-23:
The new release is not yet ready.
There are still issues open which I am working on.

73 de Martin, DL1RF

g6hfs, 2024-11-23:
Hi Martin,Thankyou, look forward to its release.73Brian (G6HFS)

dl1rf, 2024-11-23:
The dBb indicator will look like:

[image]

73 de Martin, DL1RF

DH1RK, 2026-09-30:
FYI: the signal strength relative to the beacon (dBb) is now also available in the DH1RK fork.

It is based on DL1RF's work (commit `183d1ac`, "show dBb values in BATC Spectrum"), which is in dl1rf's fork but not in upstream `main`. I took over only the dBb part: the value is shown as an extra line under callsign, frequency and symbol rate of each signal in the BATC spectrum. There is no value for the beacon itself, or as long as no beacon has been seen.

- Issue: DH1RK/open_tuner#47
- Change: DH1RK/open_tuner#48 (merged into the fork's `main`, will be part of the next beta, `V0.B-dh1rk-beta6`; not released yet)

Not done: showing it only while hovering over a signal bar, as originally asked in this issue. The value is always drawn.