tomvdb/open_tuner #4: open_tuner : Japanese version

shinji-y-jp, 2023-03-19:
Hello there

I am trying to modify open_tuner for localize Japanese version .
Please watch pics , that's my result .
We are unable to watch the fft window .
Please point out where I have to modify .

Shinji

[image]
[image]

shinji-y-jp, 2023-03-20:
Hi there

The modified point :
file        Form1.cs:
            initialConfig.frequency = 1298000;
            initialConfig.symbol_rate = 333;

file        soket.cs:
                /*
                Console.WriteLine(connected);
                Console.WriteLine("Try connect..\n");
                // System.Threading.Thread.Sleep(500);     //can't catch exception from websocket!?, slow down retries if no network

                //ws = new WebSocket("wss://eshail.batc.org.uk/wb/fft", "fft_m0dtslivetune");
                ws = new                 WebSocket("ws://192.168.0.244:7681", "fft_m0dtslivetune");
                ws.OnMessage += (ss, ee) => NewData(ee.RawData);
                ws.OnOpen += (ss, ee) => { connected = true; Console.WriteLine("Connected.\n"); };
                ws.OnClose += (ss, ee) => { connected = false; };
                ws.Connect();
                */ comment outed

Thanks in advance ,

Shinji / JE1BTA

tomvdb, 2023-03-20:
Hi Shinji,

Thanks for you messages.

(1) The fft display is not from the minitiouner hardware but from the eshail wideband monitor (https://eshail.batc.org.uk/wb/) and it requires an internet connection to work. One of the next updates will be to have a setting that it can be switched off for non eshail use. 

I have a few ideas on doing a more real fft from the hardware or maybe even with the addition of an sdr, but I want to get the basics going on open tuner first.  

If I misunderstood and you actually don't have a working fft with the internet on from the batc site then I'll need to add some more debugging to see whats going on.

(2) Out of interest, what changes do you required for japanese localization? One of the items on my list is to integrate different languages system so that we can load different translations. Is this what you had in mind, or are these specific changes required?

Looking forward to your replies,
Tom - ZR6TG

shinji-y-jp, 2023-03-20:
Hi Tom,

Thank you for your heartfelt message . I am so glad .
I understood overview of open tuner and your mind .

>what changes do you required for japanese localization?
By the way , Hiroshi JA1SYK who is a pioneer of DVB-S2 in Japan will tell you the Japanese localization detail specs and requirements , that's works well ?
Cos my English ability and understandings about DATV is poor .

>Is this what you had in mind, or are these specific changes required?
The answer is yes . Hope , open tuner prosperity and multilingual working 

With the truth ,

Shinji - JE1BTA

shinji-y-jp, 2023-03-20:
Hi Tom,

In my words , what Hiroshi saying :
At first 1200MHz band.
Selectable QO-100 mode and Terrestrial mode by button
fft windows :
Center frequency 1298MHz , span 1MHz

These are Hiroshi's requirements .

Thanks a lot ,

Shinji

tomvdb, 2023-03-22:
Hi Shinji,

I understand the requirement. I'll need to think about best way to implement something like this, but its on the todo list. I'll keep you updated,

Thanks,
Tom

shinji-y-jp, 2023-03-22:
Hi Tom ,

Thank you for your understandings . In my private opinion , as simple way temporally spec , to enlarge the receiving video windows and not showing spectrum how ?
Cos I know real time fft implementation is hard for us .

Thanks ,

Shinji

tomvdb, 2023-03-22:
If you look at the latest version you can disable the qo-100 spectrum :

[image]

and then it looks like this

[image]

Is that what you had in mind ?

shinji-y-jp, 2023-03-22:
Hi Tom ,

Perfect ! I must apologize to you .

Thanks ,

Shinji

tomvdb, 2023-04-28:
Thanks Shinji! I'll be fixing up the various languages for the next release.

73

closed 2023-04-28 by tomvdb