tomvdb/open_tuner #6: Open Tuner tested on different Minitioune hardware

lubuntu-l100, 2023-03-20:
Hi Tom,

I have tested the 19-03-2023 version of the Open Tuner software on a Minitiouner Express hardware as well on the Minitiouner S hardware. On both hardware the open Tuner software is working fine. No real issues to mention. Open tner is much faster than the minitiouner software. Changing from frequency in 4 second with Open Tuner against 7 seconds for the minitiouner software in my case. Thanks for this great work.

I have also some needs:  The text in the left side of the screen is very small and hard to read for an old man if you can change the font to 10 pitch that should be nice. Even so without connection de D value is 9.9, I should expect 0.0 because there is no signal. 

Even so the Power I nd Q values have from my opion no added value. The LDPC count should be nice is that is a % instead of continue flipping counts in the hundreds.
LNA gain is showing a value of 127 at my sight but is should be 12.7 dB. 

These are only cosmetic changes. A real wish is to change the frequency also form another pc like Quick Tune and evenso it should be nice to show the VLC stream on another pc, like what is possible in the minitiouner software.

I have compiled the main.zip and I am just discovering the code to learn from how it is constructed. I will continue my testwork.

Thanks Tom from PA2JSA  Jaap in teh Netherlands

lubuntu-l100, 2023-03-20:
> Hi Jaap,
>
> Thanks for the feedback. Good to know its working well for you. All
> noted on your visual requirements, most of them are also on my list.
> The udp port for VLC is also in the works and I'll add quick tune
> support on the list as well :)
>
> hehe, the code is messy still, I'm still cleaning it all up, but feel
> free to ask questions if you have any.
>
> 73,
> Tom

Hi Tom,

Thanks for your quick reply.

I think this project is one of the most important  developments so that we can continue to use the minitiouner hardware with new possibilities of the software.

I have 2 minitiouners in use an old mintiouner express from the USA and I have modified the hardware a littlebit. Last year I bought the components for the French minitiouner S.

Both are working fine and If you need help to test new releases on both type of hardware. No problem. Please let me know. As an pensioner I have plenty of time to test these developments.

In my professional life I was working for 20% as an associate professor at the Technical University in the Netherlands as well as at the Brussels School of Management in Belgium on the topics of Enterprise and Cyber Architectures.

The rest of my time I was worldwide responsible for Cyber Protection & Resiliance in Industrial Control Systems for a multinational located in Canada.

I have designed several Cyber Protection Architectures for the oil & gas industry, the chemical industry as well as Nato Defence forces.

So I am familiar with hardware as well as software development in several areas.

I am also QRV on Datv since summer 2019  see my qrz.com page.

Thanks Tom for your great work.

73  Jaap

PA2JSA

BlueMail voor Android downloaden

lubuntu-l100, 2023-03-21:
> Hi Jaap,
>
> Thanks for all the info. I have implemented some of the visual
> requests, including the font change for the main stats, hopefully that
> helps so long. The other changes are on the list and will happen over
> time. :)
>
> I've also added better detection for the different minitiouner variants
> and ftdi ports so could be interesting if you can test that your
> versions detect properly after the changes.
>
> Thanks,
> Tom

Hi Tom,

I will do the tests and will report you about my observations.

Thanks for the new version. I have it already installed on 2 pc's.

I have send you by paypal a contribution for your efforts.

73  Jaap

PA2JSA

BlueMail voor Android downloaden

lubuntu-l100, 2023-03-21:
Hi Tom,

 

I did some test with the Open-Tuner 21-03-2023 version on the Minitiouner Express as well as on the Minitiouner S. Below you will find my remarks.

 

Thanks for the improvements with the text in the left column, that is now good readable for me.

 

Attached you find 2 log files, 1 one from the Minitiouner Express and 1 from the Minitiouner S. These logs are from starting up the App, connecting to the device and then opening the Beacon signal. The log from the Minitiouner Express also logged the change to another signal and stopped when the signal disappeared. The logs are showing a couple of errors. However from the app perspective everything was working.

 

This morning with the Minitiouner Express, I observed a strange behavior. There was a small 125 K/s signal on the WB and I tried to see what this signal was. It was not a DVB-S signal, so the Demod State was Hunting while the signal was still there. I started the original Minitiouner software and even there the signal was available but a TS stream was not detected. Possible a DVB-GSE signal. However the Open-Tuner software locked itself up and was not reacting on the user interface anymore. So I needed to kill the app and then started again. Look like the software don’t understand what to do in such a case.

 

I discussed the open-tuner software also in our daily round at 700 on the NB at 9.30 AM in Dutch.

 

Some interesting needs and observations ware discussed:

 

*         The possibility of selecting Output A or Output B of the Nim. Default is output A, but some people use output B instead of A.

*         When connected to a Minitiouner there is No option to Disconnect. The only way is to stop the App.

*         When you close the App, but the logfile is still open and start the app again there is a message: error: No working Hardware Detected. Push on the Ok button and trying again, results in the same message. Closing the logfile and then starting again Open Tuner, the software is working fine.

*         The other points we already discussed.

 

This is it for now.

 

73’  Jaap

 

PA2JSA

 

 

Call Sign: PA2JSA

Operator: Jaap Schekkerman

Location: JO22QE

Country: The Netherlands

Email:  ***@***.*** ***@***.***>  

__________________________IMPORTANT NOTICE_______________________

The content of this email and any attachments are confidential and intended for the named recipient(s) only. If you are not the intended recipient of this e-mail, any dissemination, distribution or copying of this e-mail, and any attachments thereto, is strictly prohibited. If you have received this e-mail in error, please immediately notify the sender by e-mail or by telephone and permanently delete all copies of this e-mail and any attachments. The name of the sender at the bottom of this message or contained anywhere else in this e-mail is only for identification and is not intended to and shall not act as a signature indicating agreement or the formation of a contract unless that is specifically stated in the body of the e-mail message. 

WARNING: Although PA2JSA has taken reasonable precautions to ensure no viruses are present in this email, PA2JSA cannot accept responsibility for any loss or damage arising from the use of this email or attachments.

 

[... 181 lines cut ...]
Main: Stopping VLC
[0596d210] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 4, expected 11) for PID 256
Close Media Input
VLC: Stopped
[0593f3e0] direct3d11 vout display error: SetThumbNailClip failed: 0x800706f4
Lock State Change: False->True
Startng TS queue and VLC
Main: Starting VLC
[0596cf10] imem demux error: Invalid get/release function pointers
Open Media Input
VLC: Playing
Clearing out TS Queue
Done... starting buffering
[12ea1290] main decoder error: buffer deadlock prevented
[13240628] main decoder error: buffer deadlock prevented
[13240b78] mpeg4audio packetizer: AAC channels: 1 samplerate: 22050
[0593fe08] direct3d11 vout display error: SetThumbNailClip failed: 0x800706f4
[0593f3e0] direct3d11 vout display error: SetThumbNailClip failed: 0x800706f4
Lock State Change: True->False
Stopping TS queue and VLC
Main: Stopping VLC
TSStreamMediaInput : Read Timeout
Close Media Input
VLC: Stopped
[0593f7f0] direct3d11 vout display error: SetThumbNailClip failed: 0x800706f4

tomvdb, 2023-03-21:
Hi Jaap,

Thanks for all the info, and those logs.

> The logs are showing a couple of errors. However from the app perspective everything was working.

VLC complains about a lot of different things, but for most cases the TS packets are valid, but the encodings done on user side has errors. It displays those errors, but it recovers from most of them by themselves so should be fine, once the more basic stuff is in place then I'll investigate the various errors and see which ones I can actually do something about

> This morning with the Minitiouner Express, I observed a strange behavior. There was a small 125 K/s signal on the WB and I tried to see what this signal was. It was not a DVB-S signal, so the Demod [...]

The TS parsing is currently very basic and can definitely do with some improvement, although it shouldn't hang. I have seen similar effects though and I suspect that is actually vlc that hangs due to those packets. In theory I should probably only pass along the packets that are valid for video/audio to vlc. But more experimentation is required. It could also be not related to that at all, but rather to 125ks signals as there has been other mentions on funny things happening on 125ks. Its on the list for investigation.

> The possibility of selecting Output A or Output B of the Nim. Default is output A, but some people use output B instead of A.

I added this today and will be available in the next release.

>When you close the App, but the logfile is still open and start the app again there is a message: error: No working Hardware Detected. Push on the Ok button and trying again, results in the same message. Closing the logfile and then starting again Open Tuner, the software is working fine.

Yes, thats normal, the "log file" is actually the app still running. This is currently only temporary for easier debugging, but the console part will fall away once all the basics are in place and working

>When connected to a Minitiouner there is No option to Disconnect. The only way is to stop the App.

True, The connect button was just part of the testing initially, but at some point I figured it would fall away as the app could connect automatically when it starts up. I guess having a connect/disconnect would be handy as well. Let me think about this one.

Thanks for all the feedback :)

lubuntu-l100, 2023-03-21:
Hi Tom,

Thanks for the info. I will look for the next version and will start again testing.

I will let you know the outcome.

73  Jaap

PA2JSA

BlueMail voor Android downloaden

lubuntu-l100, 2023-03-22:
Hi Tom,

 

Thanks for version 2023/03.21b of Open Tuner.

 

I notified that you have restructured the Form1.Designerr.cs in a way that you transposed the layout content to the Form.resx. A . net file. Well done this makes it easier to modify the layout.

 

For myself I did some minor layout changes like:  Replacing Service Name by Callsign and replacing Service Name Provider by Service Description. I think that better represents in our case what it is.

 

I did several tests:

 

RF Input A is working fine. RF Input B is also working fine If you have an external BiasT. Switching from A to B is working fine however the Power of the LNB is still on A. It should handy if the Power switches from A to B in Line with the RF Input switch.

 

Comparing the RF Input Level of Input A or B, they are the same. However without a signal the RF input is around -82dB with the beacon locked it is -62dB at my side. I noticed that switching from RF Input A to B, not always the RF Input Level is refreshed. So sometimes it looks that the beacon has an input level of -82db. Again activating the beacon brings that to the original -62dB.

 

I did also some test with smaller symbol rates from my own DATV transmitter. 333K/s is ok, 250 K/s is also working fine but 125 K/s is a challenge. It took a while before there was a video but it was a little bit stuttering. The problem good by on my site because I use an external H265 encoder box and the minimum video bitlrate is 60 K/s. A symbolrate of 125K/s results in a video bitrate of around 68K/s. So that is close to the lower limits.

 

I even so noticed that the Quit function in the Action tab not always worked. A couple of times it did nothing. However it was also not reproducible. 

 

These are my tests and observations of today on this version.

 

I downloaded the source files from github for this version and earlier versions, I noticed that everything is there accept the “bandplan’ xml. This file is available in your compiled version but not in the directory of the source files. So I copied this from the Compiled.zip file.

 

Thanks for your efforts Tom.  I am learning from the source code.

 

73’  Jaap   PA2JSA

 

Call Sign: PA2JSA

Operator: Jaap Schekkerman

Location: JO22QE

Country: The Netherlands

Email:  ***@***.*** ***@***.***>  

__________________________IMPORTANT NOTICE_______________________

The content of this email and any attachments are confidential and intended for the named recipient(s) only. If you are not the intended recipient of this e-mail, any dissemination, distribution or copying of this e-mail, and any attachments thereto, is strictly prohibited. If you have received this e-mail in error, please immediately notify the sender by e-mail or by telephone and permanently delete all copies of this e-mail and any attachments. The name of the sender at the bottom of this message or contained anywhere else in this e-mail is only for identification and is not intended to and shall not act as a signature indicating agreement or the formation of a contract unless that is specifically stated in the body of the e-mail message. 

WARNING: Although PA2JSA has taken reasonable precautions to ensure no viruses are present in this email, PA2JSA cannot accept responsibility for any loss or damage arising from the use of this email or attachments.

tomvdb, 2023-03-22:
Hi Jaap,

Thanks for all the feedback. Some comments from my side:

>I notified that you have restructured the Form1.Designerr.cs in a way that you transposed the layout content to the Form.resx. A . net file. Well done this makes it easier to modify the layout.

This is part of the multi language setup. I've just pushed code that adds various languages.

>For myself I did some minor layout changes like:  Replacing Service Name by Callsign and replacing Service Name Provider by Service Description. I think that better represents in our case what it is.

The TS packet where I get my info is service packet and those values comes from the TS spec (which obviously doesn't really cater for amateur radio).

>RF Input A is working fine. RF Input B is also working fine If you have an external BiasT. Switching from A to B is working fine however the Power of the LNB is still on A. It should handy if the Power switches from A to B in Line with the RF Input switch.

Thats interesting. Everything on my side uses Bias-T so I don't use any of the lnb power functions. A good test would be to reset the power after the input has been changed? ie. change to B, then switch the lnb supply setting to off and then to horizontal again. It might need to be reset when the rf input is changed.

> Comparing the RF Input Level of Input A or B, they are the same. However without a signal the RF input is around -82dB with the beacon locked it is -62dB at my side. I noticed that switching from RF [...]

I haven't noticed this on my side. I will need to investigate.

> I did also some test with smaller symbol rates from my own DATV transmitter. 333K/s is ok, 250 K/s is also working fine but 125 K/s is a challenge. It took a while before there was a video but it [...]

So, I've been doing some more reading and this comes down to how the tuner is configured and the register sizes. ie, its a bit more tricky than what it sounds and also requires some calibration on the fly. I'm sure Minitioune does some interesting tricks to achieve the lower symbol rates. I'll get there, but it will still take some time ;)

>I even so noticed that the Quit function in the Action tab not always worked. A couple of times it did nothing. However it was also not reproducible.

Yeah, I got that today as well. The threads that are running doesn't immediately shutdown when they get the signal to abort. Its on the list, but its a lower priority.

>I downloaded the source files from github for this version and earlier versions, I noticed that everything is there accept the “bandplan’ xml. This file is available in your compiled version but not in the directory of the source files. So I copied this from the Compiled.zip file.

Yeah, the bandplan file is in my binary folder which is in the .gitignore so not going onto github repo. I've added an extra folder for files like that.

btw - I'll have a new version out tonight which also runs on much less cpu load so keen to hear your results on the new version.

Thanks,
Tom

tomvdb, 2023-03-22:
Hi Jaap,

I just remembered that the lnb power supply switching is seperate from the nim module, so that shouldnt be affected by which input is selected, but rather depends on the lnb power circuitry. I'll check it out tonight, but chances are I'm not switching it properly. Will revert.

T

lubuntu-l100, 2023-03-22:
Hi Tom,

Thanks for the feedback. Lets see what the problem is.

Thanks.

73  Jaap PA2JSA

BlueMail voor Android downloaden

lubuntu-l100, 2023-03-23:
Hi Tom,

 

Below my observations and remarks on version 2023-03-22.

 

Great to see that the cpu usage has been dramatically lowered, in my case from 35% down to less than 10%. I have tested first the Multilanguage options. I noticed that by default, the software is looking at the windows default language. So I have English as well Dutch on my laptop. Changing the language in Windows changes also the language in Open Tuner. When my default language is English I gain change the language in Open Tuner to Dutch and after a restart the language Dutch is there. When my default language is Dutch, I cannot select English in Open Tuner, because English is not in the selection window. It is a nice to have that also in the selection window.

 

After that I tested the software in several ways and Identified problems with VLC after switching to RF input B and back. With Rf input B most of the time I can select a station but vlc is not starting, switching back to rf input A, data and VLC are not inline. In the data site is the data of the beacon while vlc is showing another station.

 

See the image below.

 

 

With the earlier versions of Open Tuner I did the same type of test and that was working fine.

 

I even so noticed that when switching to RF input B and clicking on a station. An invalid frequency message in red appears in the Tuner control window. This was reproducible on the Minitiouner Express as well as the Minitiouner S.

 

I did the same tests on the Minitiouner S and there appears a message of Hardware Error, while there was no hardware error. Starting up with the French Minitiouner software everything was working fine. Back to open tuner and that was also working. So somewhere the open tuner software appears with this message see logfile S3.

 

Attached you will find a series of log files with the different messages.

 

Thanks for all your efforts Tom.

 

73   Jaap   PA2JSA

 

 

 

Call Sign: PA2JSA

Operator: Jaap Schekkerman

Location: JO22QE

Country: The Netherlands

Email:  ***@***.*** ***@***.***>  

__________________________IMPORTANT NOTICE_______________________

The content of this email and any attachments are confidential and intended for the named recipient(s) only. If you are not the intended recipient of this e-mail, any dissemination, distribution or copying of this e-mail, and any attachments thereto, is strictly prohibited. If you have received this e-mail in error, please immediately notify the sender by e-mail or by telephone and permanently delete all copies of this e-mail and any attachments. The name of the sender at the bottom of this message or contained anywhere else in this e-mail is only for identification and is not intended to and shall not act as a signature indicating agreement or the formation of a contract unless that is specifically stated in the body of the e-mail message. 

WARNING: Although PA2JSA has taken reasonable precautions to ensure no viruses are present in this email, PA2JSA cannot accept responsibility for any loss or damage arising from the use of this email or attachments.

 

System Culture Setting: en-US

[... 571 lines cut ...]
[05eadb58] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 0, expected 1) for PID 256
[05eadb58] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 0, expected 1) for PID 256
[05eadb58] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 0, expected 1) for PID 256
[05eadb58] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 0, expected 1) for PID 256
[05eadb58] ts demux error: libdvbpsi error (misc PSI): Bad CRC_32 table 0x89 !!!
[05eadb58] ts demux error: libdvbpsi error (PSI decoder): PSI section too long
[05eadb58] ts demux error: libdvbpsi error (PSI decoder): PSI section too long
[05eadb58] ts demux error: libdvbpsi error (misc PSI): Bad CRC_32 table 0xc5 !!!
[05eadb58] ts demux error: libdvbpsi error (PSI decoder): PSI section too long
[05eadb58] ts demux error: libdvbpsi error (PSI decoder): PSI section too long
[05eadb58] ts demux error: libdvbpsi error (PSI decoder): PSI section too long
[05eadb58] ts demux error: libdvbpsi error (PSI decoder): PSI section too long
[05eadb58] ts demux error: libdvbpsi error (PSI decoder): PSI section too long
[05eadb58] ts demux error: libdvbpsi error (PSI decoder): PSI section too long
[05eadb58] ts demux error: libdvbpsi error (misc PSI): Bad CRC_32 table 0xfb !!!
[05eadb58] ts demux error: libdvbpsi error (PSI decoder): PSI section too long
[05eadb58] ts demux error: libdvbpsi error (PSI decoder): PSI section too long
[05eadb58] ts demux error: libdvbpsi error (PSI decoder): PSI section too long
[05eadb58] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 14, expected 15) for PID 256
[05eadb58] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 14, expected 15) for PID 256
[05eadb58] ts demux error: libdvbpsi error (PSI decoder): TS discontinuity (received 14, expected 15) for PID 256
[05eadb58] ts demux error: libdvbpsi error (PSI decoder): PSI section too long
[05eadb58] ts demux error: libdvbpsi error (misc PSI): Bad CRC_32 table 0xb6 !!!
[05eadb58] ts demux error: libdvbpsi error (PSI decoder): PSI section too long
[05eadb58] ts demux error

lubuntu-l100, 2023-03-23:
Hi Tom,

 

For  your information.

 

I did another test with the Minitiouner Express as well as the Minitiouner S on the same old laptop (I3 from 2012).

 

And yes I can run both instances of Open-Tuner side by side without any problems. Of course the cpu usage is much higher than 1 instances but together they consume about 36% of the cpu and that’s acceptable.

 

73  Jaap PA2JSA

 

 

Call Sign: PA2JSA

Operator: Jaap Schekkerman

Location: JO22QE

Country: The Netherlands

Email:  ***@***.*** ***@***.***>  

__________________________IMPORTANT NOTICE_______________________

The content of this email and any attachments are confidential and intended for the named recipient(s) only. If you are not the intended recipient of this e-mail, any dissemination, distribution or copying of this e-mail, and any attachments thereto, is strictly prohibited. If you have received this e-mail in error, please immediately notify the sender by e-mail or by telephone and permanently delete all copies of this e-mail and any attachments. The name of the sender at the bottom of this message or contained anywhere else in this e-mail is only for identification and is not intended to and shall not act as a signature indicating agreement or the formation of a contract unless that is specifically stated in the body of the e-mail message. 

WARNING: Although PA2JSA has taken reasonable precautions to ensure no viruses are present in this email, PA2JSA cannot accept responsibility for any loss or damage arising from the use of this email or attachments.

tomvdb, 2023-03-23:
Hi Jaap,

>I even so noticed that when switching to RF input B and clicking on a station. An invalid frequency message in red appears in the Tuner control window. This was reproducible on the Minitiouner Express as well as the Minitiouner S.

Yes, this adds up. I've added an offset configuration for either A/B. Most likely your offset setting for B is still defaulted to 0, so when you switch to B the offset is 0, but by clicking on the spectrum you are setting a frequency in the 10ghz range, so the resulting frequency is out of range for the minitiouner.

Regarding the power on the lnb ports issue. The original longmynd only made provision for one lnb enable output on the ftdi. Based on this wiki entry (https://wiki.batc.org.uk/Serit_LNB_DC_supply#LNB_power_supply_with_current_limit_for_MiniTiouner) it seems to indicate that for the batc minitiouner both enable lines for the two lnb powers get connected to that specific output so they will both switch as needed. But on the minitiouner pro schematic there are two different outputs to be able to switch each of them independently. I can make provision for this, so its not an issue, but it explains why the 2nd one is getting power. The question is whether the Express and S versions of minitiouner make provision for lnb power switching ? Can you maybe check that on yours? I suspect from what I've read that they don't (or atleast the express) doesn't have any circuitry for lnb power switching using the RT5047

Multiple instances of opentuner should be fine, but most likely will not work with two of the same hardware types as it will probably try to open up them both and fail on the one, but using for example an express and s should be fine, because they are detected seperately. Part of the changes I want to do is to support both tuners on the pro2, so that can maybe be smart enough to deal with multiple minitiouners in the same app, but I need to get my hands on more hardware first to get that going.

73,
Tom

lubuntu-l100, 2023-03-23:
Hi Tom,

Thanks for your quick reply.

Both the Minitiouner Express as well as the French Minitiouner S are using an external biasT. So myself I don't have problems.

But the French Minitiouner v2 Pro have a powermodule for the lnb onboard. PA1RW has such a model. He reported the problem with the power on rf input B.

You did not mention the problems with vlc that frequently reported errors in the log, see my attached logfiles on github.

Probably that problem is still under investigation.

Thanks for your efforts.

73  Jaap PA2JSA

BlueMail voor Android downloaden

lubuntu-l100, 2023-04-02:
Hi Tom,

 

I tested the new Open Tuner software version 04-01-2023 and have the
following remarks / wishes:

 

*         Over power is working fine but very dominant in the screen. The
red color a little bit more transparent should be fine.

*         I changed the UDP address and Port in TSUDPThread.cs and compiled
the software and the change of UDP IP address as well as Port address is
working fine. I found no possibility to change the address and port within
the open tuner software itself. But it is working fine to stream the output
to another PC.

*         I tried in several ways the opentuner.exe and autoconnect but was
not successful. Probably the syntax I used is incorrect. Please let me know
the correct syntax.

*         Aspect Ratio switch is working fine.

*         I like to see the possibility to select a station in the quick
interface from another pc where the standalone quicktune is running or from
the Pluto.php interface I use with the patch IS0GRB v2.1. I can tune the
minitiouner on another PC from this Pluto.php. I like to see the same
functionality in Open Tuner. The Pluto.php or quicktune standalone is
sending the selected frequency over UDP to 232.0.0.11 and port 6789 to the
minitiouner receiver. It should be nice as open tuner can also receive this
information over UDP 232.0.0.11 and port 6789.

*         It should be fine in the program settings to do the selection
Enable UDP output by default instead of activating it each time you start
the program by yourself.

 

Tom, thanks for your great work.

 

73'   Jaap   PA2JSA

 

 

Call Sign: PA2JSA

Operator: Jaap Schekkerman

Location: JO22QE

Country: The Netherlands

Email:   ***@***.***> ***@***.*** 

__________________________IMPORTANT NOTICE_______________________

The content of this email and any attachments are confidential and intended
for the named recipient(s) only. If you are not the intended recipient of
this e-mail, any dissemination, distribution or copying of this e-mail, and
any attachments thereto, is strictly prohibited. If you have received this
e-mail in error, please immediately notify the sender by e-mail or by
telephone and permanently delete all copies of this e-mail and any
attachments. The name of the sender at the bottom of this message or
contained anywhere else in this e-mail is only for identification and is not
intended to and shall not act as a signature indicating agreement or the
formation of a contract unless that is specifically stated in the body of
the e-mail message. 

WARNING: Although PA2JSA has taken reasonable precautions to ensure no
viruses are present in this email, PA2JSA cannot accept responsibility for
any loss or damage arising from the use of this email or attachments.

tomvdb, 2023-04-04:
Hi Jaap,

Thanks for your feedback as always.

(1) Yes, I'm still playing with a visual indication on the over power. Stay tuned :)
(2) The UDP will be part of the settings in the next version. I wasn't sure how well it was going to work, but complaints have been minimal, so I can lock it down a bit better now.
(3) for the autoconnect your run "opentuner.exe AUTOCONNECT", I need to spend time on better documentation ;)
(4) good point on the Enable UDP checkbox, will add to settings
(5) I'll be adding quicktune control facility as well, probably next release 

Thanks,
T

lubuntu-l100, 2023-04-04:
Hi Tom,

Thanks for your reply.

Vlc udp stream is working very well on other computers. I have noticed no problems with it.

Remote controlling open tuner via Quicktune or pluto.php should be great.

I saw some remarks about the overpower indication that it is not always working correct.

I checked the source files and noticed in signal.cs the formula for calculating the over power level. I made a small correction and that was working a littlebit better. However when the beacon is there and 2 station with 500k/s are transmitting with a power of 1 db below the beacon everything is ok. However when a 1 m/s signal starts transmitting besides the other 2, with a power also 1 db below the beacon, the two 500K/s signals got the overpower message, while they are still 1 db below the beacon. Somewhere when more concurrent signals share the WB transponder, and each other get a lower signal, the formula is not always calculating the over power value correct. Not an important issue, but sharing the observations.

73  Jaap   PA2JSA

⁣BlueMail voor Android downloaden ​

lubuntu-l100, 2023-04-05:
Hi Tom,

 

I did some further research about the over-power indication, based on comments of other HAM’s and what is happening you can see in the following pictures. When a 1 M/s signal is direct besides the beacon, Open Tuner take the signal strength of the 1M/s besides of the beacon as the reference for the calculation of overpower instead of the signal strength of the beacon itself. And so over power is indicated on a 500 K/s signal while it is NOT over powered.  Hope this info will help you.

 

 

 

 

73’  Jaap  PA2JSA

lubuntu-l100, 2023-04-05:
Hi Tom,

 

Another picture that is representative for the problem with the  over power indication.

 

 

73  Jaap  PA2JSA

lubuntu-l100, 2023-04-08:
Hi Tom,

 

I did some further investigation in the source files of open tuner and I
noticed in Signal.cs Line 490 the following syntax:

 

490          if (signal_freq < 10492000 && signal_bw >=  1.0) I read this
that as the frequency is below 10492000 and the symbol rate is greater or
equal to 1M/s the   beacon_strength = strength_signal;

 

I changed that line 490 in 

490          if (signal_freq < 10492000 && signal_bw > 1.0) so that only
when the frequency is below 10492000 and the symbol rate is greater than 1
M/s

beacon_strength = strength_signal;

 

So excluding 1M/s signals in the case your tuner is not on frequency which
can be the case due to different  LNB's.

 

I compiled the software and did some test over the last days. The strange
behavior was not any more there.

 

Below a picture with the changed source code and an over-power situation
with the modified Signal.cs.

 

 

 

 

It look likes that this is a possible solution for the strange over power
behavior.

 

73'   Jaap

PA2JSA

tomvdb, 2023-04-13:
Hi Jaap, Thanks for all the information and digging regarding the over power issues on the spectrum. Unfortunately none of your pictures you sent is coming through to the github issues platform. Do you mind emailing me directly with the pics so that I can check it out ?

Thanks,
Tom

lubuntu-l100, 2023-04-13:
Hi Tom,

Probably it is easier to send you the pictures directly. Send you email addres and I will send you the info directly.

With the modified signal.cs file, it look likes that the over power indication is working correct. Did some tests Saterday at the same time that Gary was testing its pa with 1Ms to set it 1 db lower than the beacon. That was interesting to see that the over power indication was going on and of. So I think with the modified signal.cs over power is working fine.

73  Jaap  PA2JSA

⁣BlueMail voor Android downloaden ​

tomvdb, 2023-04-28:
Hi Jaap, i'm closing off this issue on github as its turned into a bit of thread and I'm cleaning up the next issue/todo list. If there is anything individual I missed then lets create seperate issues for those.

Thanks for all the feedback!
73

closed 2023-04-28 by tomvdb

lubuntu-l100, 2023-04-28:
Ok Tom,

No problem. Thanks for your feedback.

73. Jaap

PA2JSA

⁣BlueMail voor Android downloaden ​