tomvdb/open_tuner #11: Device Detection MiniTiouner-Express and -S

JayCeeSharp, 2023-03-21:
Hi Tom,
tnx for very fast response (rightclick TX-freq.)!

I found this devicenames corresponding to my tiouners
73
Chris

changes to ftdi.cs:

                    if (deviceName.Contains("MiniTiouner A"))
                    {
                        Console.WriteLine("Should be the i2c port for a Minitiouner Express"); // dh3cs seems to be not 
                        i2c_port = c;
                        detectedDeviceName = "Minitiouner-S ";   // dh3cs  was "Minitiouner Express";
                    }

                    if (deviceName.Contains("MiniTiouner B") || deviceName.Contains("Express"))   //dh3cs MiniTiouner-Express A
                    {
                        Console.WriteLine("Should be the ts port for a Minitiouner Express");
                        ts_port = c;
                        detectedDeviceName = "MiniTiouner-Express";  // dh3cs added line
                    }

tomvdb, 2023-03-21:
Based on the logs from Jaap (https://github.com/tomvdb/open_tuner/issues/6), it seems to be as follow:

MiniTiouner A
MiniTiouner B

MiniTiouner-Express A
MiniTiouner-Express B

I'll update accordingly, thanks Chris

JayCeeSharp, 2023-03-21:
in the meantime I updated the comment - that's correct.
I was a little bit confused,while both devices connected at the same time

tomvdb, 2023-03-22:
This should be resolved in the latest version, we can re-open if still an issues. Thanks for the feedback :)

T

closed 2023-03-22 by tomvdb