tomvdb/open_tuner #12: Opentuner Nim Select

g6hfs, 2023-03-21:
Tom,

Wanted to change Nim A or B input, could this be an option please

Brian (G6HFS)

tomvdb, 2023-03-22:
Hi Brian,

As you saw the latest version has the A/B selection. I can't seem to reproduce your issue though, does it just close completely whenever you try to switch between the two? Is there an error of sorts? Does it happen every time, or just occasionally ?

Any more info you can provide will help me troubleshoot this.

Thanks,
Tom

g6hfs, 2023-03-22:
Tried to send video

g6hfs, 2023-03-22:
No luck

tomvdb, 2023-03-22:
Hi Brian,

Thanks for that. I'll need to setup a test on my bench to try and replicate it, the difference being that you have two physically different signals to test with, and im just swopping the single cable. Let me play a bit and see what I can find.

T

g6hfs, 2023-03-22:
Tom,

Yes it only seems to happen after using the manual tuner controls.

Thanks for your great hard work

Brian

tomvdb, 2023-03-22:
Hi Brian,

I think i've located this issue. You mention that you change the offset when switching to B, but when you switch back do you change offset again back before selecting the beacon on the spectrum ? There is no error checking on the frequency, so lets say for example you changed frequency to something in the 70cm range, and you changed you offset to 0, then when you switch back to A and select the beacon it will try to configure the tuner with the beacon frequency (10491520), but you offset is still on 0 so it tries to configure the tuner with that high frequency. This fails and because there is no error checking it down right crashes. I've added better error checking and also added support from setting a default offset for A and B so that it will automatically update the offset depending on which rf input selection you are.

I hope the above make sense. I'll email you a test version in a bit to see if that works better for you.

73,
Tom

g6hfs, 2023-04-28:
No problem with the new 0.7 release Tom

tomvdb, 2023-04-28:
Thanks Brian!

closed 2023-04-28 by tomvdb