tomvdb/open_tuner #8: Translations Requested

tomvdb, 2023-03-21:
Request for language translations:

I would like to have multi languages in Open Tuner. If anybody is keen to help translate then please have a look here:

https://docs.google.com/spreadsheets/d/1KZA6D-Fmkw39m_3Otmb1MuGoiDdZvLq2ol81NF_4NMM/edit?usp=sharing

The google doc contains words/expressions/etc that needs translating. Simply make a copy of the spreadsheet, add in the translations and send/share with me. 

Thanks,
Tom

shinji-y-jp, 2023-03-21:
Hi Tom

I do . I am a native Japanese . Japanese language consists of  Hiragana , Katakana , and Kanji ( which means Chinese character ) . So I would translate them to Japanese language .

Is that works well ? If so I do .

Sincerely ,

Shinji - JE1BTA

tomvdb, 2023-03-22:
Hi Shinji,

Yes, that would definitely be helpful. It would be a good test to see if the app can work with the characters. Can you copy the spreadsheet linked above and add the translations into the 2nd column and send to me, then I'll work it in.

Thanks,
Tom

shinji-y-jp, 2023-03-22:
Hi Tom ,

Here it is :
https://docs.google.com/spreadsheets/d/1Ur0gL8G5uqImQbaCs3Job9FzBishMHgMORrwJgoGN8c/edit#gid=0

I would not know the items function . So in the future , after translations completed , in case I felt strange translation , they will change .

Thanks ,

Shinji

tomvdb, 2023-03-22:
I currently have the following translations:

Japanese - JE1BTA
Italian - I4WRM
Dutch - PA3FBX
German - DH3CS

Thanks all

Jabieraru142, 2023-03-22:
Dear Tom: I am ready to translate to spanish. I send you news. 73,s de Jabi, ea2aru.

tomvdb, 2023-03-23:
Spanish and Polish translation added into the code, will be part of next release. Thanks to EA2ARU and SP9ACQ

F1EJP, 2023-04-24:
Hello Tom

Here is in French for your fantastic soft :)
https://docs.google.com/spreadsheets/d/1ti7kdnbE86x1s50ZO2nXJ3VJh2tFQgQf8jvP6ur-Bk0/edit?usp=sharing

73 Dominique F1EJP

tomvdb, 2023-04-27:
> Hello Tom
> 
> Here is in French for your fantastic soft :) https://docs.google.com/spreadsheets/d/1ti7kdnbE86x1s50ZO2nXJ3VJh2tFQgQf8jvP6ur-Bk0/edit?usp=sharing
> 
> 73 Dominique F1EJP

Thanks Dominique. Currently with all the recent changes the languages are a bit all over the place. After the next release I'm going to spend some time on all the received languages and get it updated.

Thank you!