tomvdb/open_tuner #9: Indication of RF input Level Requested

davecrump, 2023-03-21:
Ryde and Portsdown (and MiniTioune) display the RF input level to the tuner in as a figure in dB that can be used to check receiver gains and that the receiver is operating with an optimum input level.  The figure is not dBm, but is dB linear and is generally within +/- 15 dB of dBm.  It is consistent across all receiver softwares.

The RF input level is calculated from the AGC1 and AGC2 values output by the latest version of LongMynd (ID 26 and 27 here [https://github.com/BritishAmateurTelevisionClub/longmynd#status-output-interface](url) ) and converted to the dB figure in a look-up table in Ryde and Portsdown which is tabulated here: [https://wiki.batc.org.uk/MiniTiouner_Power_Level_Indication#Tabulated_Results](url).

It would be useful to have this RF input level figure displayed in Open Tuner rather than the Power I and Power Q figures which are meaningless to most users.

Thanks!  Dave

tomvdb, 2023-03-21:
Hi Dave,

This has been added, will be live in the next release.

T

davecrump, 2023-03-21:
That was quick!  Thanks Tom

closed 2023-03-22 by tomvdb