tomvdb/open_tuner #27: DVB-S reporting

G3GJA, 2023-03-27:
D report does not work with DVB-S.

OK with DVB-S2

Clive G3GJA

tomvdb, 2023-03-29:
Hi Clive, 

I will check it out. 

T

tomvdb, 2023-03-31:
Hi Clive, 

I'm not seeing any obvious issues in the code around this. Can you send me a screenshot of where its locking on a DVB-S signal, but not showing the D, maybe there is a clue on there. I'll have to setup something to test a dvb-s signal, will try it out over the weekend.

T

PA0RWE, 2023-04-16:
Hi,
Yes I found the same problem with DVB-S (from my Portsdown.) Also 125k decoding doesn't seem to work. I tested it on different frequencies but none of the 125K worked. However, lower and higher symbol rates are no problem.

PA0RWE, 2023-04-16:
Here is a screenshot with no D info.
[image]

PA0RWE, 2023-04-17:
Hi,  the screen trying to read 125Ks.

[image]

tomvdb, 2023-04-25:
> Hi Clive,
> 
> I'm not seeing any obvious issues in the code around this. Can you send me a screenshot of where its locking on a DVB-S signal, but not showing the D, maybe there is a clue on there. I'll have to setup something to test a dvb-s signal, will try it out over the weekend.
> 
> T

Hi Clive, I have found the problem and it will be resolved in the new version :)

T

tomvdb, 2023-04-25:
> Hi, the screen trying to read 125Ks.
> 
> [image]

Thanks for the screenshot. I will look into it, but please create a new issue the 125ks problem, otherwise it will get lost :)

Thanks for the feedback,
73
Tom

tomvdb, 2023-04-28:
The reporting on DVB-S should be resolved now on latest version - 0.7 (2023/04/28) Let me know if not correct.

closed 2023-04-28 by tomvdb

PA0RWE, 2023-04-28:
Yes, it's OK now. Thanks!