tomvdb/open_tuner #28: Support Other LO frequencies?

G8YTZ, 2023-03-28:
Hi Tom,

the software works great on my NUC8i7. One thing, would it be possible to add support for other LO frequencies and add some presets? I set the LO to 5150000 for the C-Band LNB I have and it did not like it, zero IF and entering the LO frequency is fine though. Even for QO-100 I want to experiment with a broadband (SKY-Q) LNB.

Otherwise, really impressed, stable and worked first time!

Justin.

tomvdb, 2023-03-29:
Hi Justin,

Presets are on the list, should be ready for the next release. For the LO issue, are you getting a specific error? Can you show me a log, or screenshot that will explain the issue?

T

tomvdb, 2023-03-31:
Implemented in latest code, wil be part of the release this weekend :)

closed 2023-03-31 by tomvdb

G8YTZ, 2023-03-31:
That’s fantastic Tom. I look forward to downloading it and trying it out.Justin Sent from my Remington Rand UNIVACwww.g8ytz.com

G8YTZ, 2023-04-01:
Tom, on the 2023/04/01 version there is still an error when setting LO to 5150000 and Rx Frequency to 3404000 and SR to 2000 - Invalid:4293221296

If I set to IF = 0 and Frequency = 1746000 (IF) Receives ok, good MER, but no Video (Tx is sending H.264)

Justin.