tomvdb/open_tuner #31: Size of Menu-Bar and Size of Spectrum are not remembered

DL2GHM, 2023-04-01:
Dear Tom,

thanks for implementing the size and position variables.
still missing the menu size (0) and Spectrum size...

Best regards
Martin

DL2GHM, 2023-04-04:
Thank you very much. That would be great!
Best regards
Martin

dogastus, 2023-04-07:
Would it be possible to translate the text in the wideband chat? It's often in German and it would be useful to understand what they are saying without having to cut and paste into Google translate.

DL2GHM, 2023-04-07:
Hi Nigel,

that is a good idea! And it is worth to make it to its own issue (better not to mix different ideas - it is easier to separate them because they would be easier to close them one by another...)

dogastus, 2023-04-07:
> Hi Nigel,
> 
> that is a good idea! And it is worth to make it to its own issue (better not to mix different ideas - it is easier to separate them because they would be easier to close them one by another...)

Sorry, I don't use Github very often. I've created a new topic now and will delete the entry here.

tomvdb, 2023-05-01:
The splitter distance for the tuner/menu bar is now part of the settings and will be saved same as window width/height sizes when app is closed. Will be in next version :)

closed 2023-05-01 by tomvdb