tomvdb/open_tuner #33: Translate Wideband Chat text?

dogastus, 2023-04-07:
Hi Tom. Would it be possible to translate the text in the wideband chat? It's often in German and it would be useful to understand what they are saying without having to cut and paste into Google translate.
Best regards,
Nigel G8AYC

DL2GHM, 2023-04-07:
Hey Nigel,

that is a good idea!
Please add also french into german ;-) if there a kind of plugin exists...

Best regards
Marty

DL2GHM, 2023-04-07:
And till then - with translate.google.com it is possible to translate whole websites :-)
https://eshail-batc-org-uk.translate.goog/wb/chat/?_x_tr_sl=de&_x_tr_tl=en&_x_tr_hl=de&_x_tr_pto=wapp

tomvdb, 2023-04-25:
Good idea, problem is that there are different languages sometimes in the same chat. But its a fun idea and I will definitely play around with it :)