tomvdb/open_tuner #37: FFPLAY and a litle bug UDP

F1EJP, 2023-04-23:
Hello Tom,

If you manage to integrate FFplay into Open Tuner I could send you a version that decode H266 VVC !

To have the UDP stream, you need to uncheck and check the box "Enable UDP output" after receiving the data

Is it possible to adjust the BaseBand gain like in Minitioune ? 

73 Dominique F1EJP

tomvdb, 2023-04-25:
Hi Dominique,

I have integrated ffmpeg into OT. I'm not using ffplay/ffmpeg directly, but rather the shared libraries. If you can compile a version of ffmpeg (x64, v6) as shared libraries then I can definitely include that. I'm currently using the latest version of this: ffmpeg-n6.0-latest-win64-lgpl-shared-6.0.zip available at https://github.com/BtbN/FFmpeg-Builds/releases/tag/latest

I'll look into the udp and baseband gain

Thanks for the feedback,
73,
Tom

tomvdb, 2023-04-26:
H266 ffmpeg decode support added for next release. Thanks to Michael DL5OCD for the updated ffmpeg libraries.

F1EJP, 2023-04-26:
if Michael provided you with the files you no longer need me to compile?
73
Dominique F1EJP

tomvdb, 2023-04-27:
> if Michael provided you with the files you no longer need me to compile? 73 Dominique F1EJP

Doesn't look like its required no, unless you want me to try a different version? Thanks!

tomvdb, 2023-05-04:
(1) Latest version seems to work much better on ffmpeg, and h266.
(2) Made some improvements to the udp outputs, should be better now as well.

Closing issue, feel free to make a new one if any of the above persists.

73,
Tom

closed 2023-05-04 by tomvdb