tomvdb/open_tuner #39: Tuner Input B cannot be selected.

G8YTZ, 2023-04-27:
New issue for v0.7

In setup both A & B are set to zero offset.

In Memory:
Freq = 1746000
Offset = 0
Symbol Rate = 2000
RF Input = B
Default Tuner = 1

As it's a strong signal there is bleed-over between the inputs, selecting the wrong input (A) reception is possible with a weaker signal.

Previous version was ok, Minitioune is ok (just tested to confirm it is not a Minitiouner issue)

g6hfs, 2023-04-28:
No problem with Minitiouner Pro 2

G8YTZ, 2023-04-28:
Interesting. Mine is the BATC Minitiouner. Previous version was fine, so something has changed.

G8YTZ, 2023-04-28:
Tom, here is a video link (Unlisted YouTube) From the time of writing 10:13 UTC  it will take 60 minutes or so to process to 4k.

[https://youtu.be/sKqBP1yjdwQ](url)

g6hfs, 2023-04-28:
Strange, pro 2 ok even ffmpeg with vvch266

[image]

G8YTZ, 2023-04-28:
That's with RF Input A though. The issue is with Input B

g6hfs, 2023-04-28:
Sorry wrong picture, now with A and B on local repeater 

[image]

G8YTZ, 2023-04-28:
The video link above is active now, best select 4k. I'm using a single Tuner Minitiouner without second TS board, perhaps you're using something else?

G8YTZ, 2023-04-28:
Brian,

thanks for confirming my cognitive health! :-)

Justin

tomvdb, 2023-04-29:
Thanks guys, I managed to trace down this issue. I will be sorted in the next version.

G8YTZ, 2023-04-29:
Tom, that’s great! I’ll test tomorrow and report back.
Justin

g6hfs, 2023-04-29:
Has Tom released the next version?

g6hfs, 2023-04-30:
Yes, works OK here on minitiouner v2 from the batc

Brian

G8YTZ, 2023-04-30:
Hi Tom, I can confirm that Input A & B switching now works with the Minitiouner V2, however there is an issue again with the IF Offset that was fixed in an earlier release (01/04/23) e.g. Rx 3404000, LO 5150000 No reception. It might be that you need to specify a +ve or =ve offset. I'll post a new issue.
Basically I currently use 3 LOs
9750MHz - QO100
5150MHz - C-Band LNB for 9cm
929MHz  - IC-9700 23cm Ist IF when using Radio Analog board PTRX9700 for 23cm reception

tomvdb, 2023-04-30:
Cool, I'll look into it. Latest version 7c is now on website with Input fix.

closed 2023-04-30 by tomvdb