tomvdb/open_tuner #40: Have to close then reopen OT to decode H266

dead-fets, 2023-04-27:
I am seeing an issue where if I open the current OT and click on an H266 signal it locks and decodes. If I then select another H266 signal it may lock, but it will not decode. Closing OT and reopening it allows that signal to reliably lock and decode it.

 It appears repeatable. I am using a single tuner, BATC V 2.0   The OS in Win 10 Pro. The issue seems only with H266 signals. I have ffmpeg selected first, then VLC below. Thanks Tom.

g6hfs, 2023-04-28:
No problem with Minitiouner Pro 2

tomvdb, 2023-04-28:
I have picked this up as well, it seems to happen on and off. 

Two possibilities: 
For the most part i'm blindly feeding the transport data as received from the NIM to the decoders (VLC or FFMPEG). I've noticed that it sometimes tend to still have transport data from a previous signal in a buffer and neither VLC and FFMPEG likes that at all. One of the next updates will be to import the TS parser to get some more information from it and possibly ensure that each new signal gets clean data.

The alternative is that there is a problem somewhere with the ffmpeg libraries (they are custom patched for h266) so its a possibility, will need to do more testing :)

Thanks for the feedback!

Chris-Wilson123, 2023-04-28:
Hi Tom, I know nothing about coding, but when you mention still having old data something has come to mind which may or may not help. When TX'ing if I start a new session I often see a tiny snippet of the OLD video sent, just for a second, as if it's cached in some sort of buffer and not flushed out . Probably totally irrelevant to OT but I thought I'd mention it. Good luck with the development and thank you. Chris 2E0ILY

Thanks for comment G6HFS.

tomvdb, 2023-04-30:
Hi All, please try new version 7c on the website :)

dead-fets, 2023-04-30:
For me this seems addressed in version 7c, thanks Tom. I do *ON MY MACHINE* have a seemingly unresolvable issue with OT when using FFMPEG on all of H264 / H265 / H266 in that the audio from all stations starts very broken, sometimes improves over about one minute, but always remains raspy and rough sounding. VLC and especially UDP out to MVP resolves to perfect, stutter free, clean sounding audio. It may well be someting in my setup, but it has persisted and is repeatabe.

tomvdb, 2023-05-01:
I'm going to reopen this issue, because I want to figure out this ffmpeg problem. If it plays via mvp then no reason it can't play via OT. I have made a few improvements so would love to get feedback on how this version affects your sound 

https://www.dropbox.com/s/lvbqzjlm9rvou68/open_tuner_0.7d_20230501.zip?dl=0

Chris-Wilson123, 2023-05-01:
Yes, I was perhaps too hasty, I am now seeing the same issue again, although it doesn't seem as chronic.

tomvdb, 2023-05-04:
Most of the ffmpeg issues should be resolved in the current 7e version. Feel free to open a new ticket on any specific issues or if it still gives problems.

73,
Tom

closed 2023-05-04 by tomvdb