tomvdb/open_tuner #41: DVB-S  125K not working

PA0RWE, 2023-04-28:
Hi, 125K is not working on DVB-S. In Minitiouner no problem.
[image]

tomvdb, 2023-05-04:
I don't currently have anything setup for dvb-s transmit, but will look at setting up a test setup on the bench to test this. Will keep you updated.

PA0RWE, 2023-05-06:
OK Tom, I found that 100K is working well.....
No hurry ;-)
Rob PA0RWE

tomvdb, 2024-07-10:
Can you try the latest version please and let me know if same issue persists please

https://www.dropbox.com/s/2dcg7eq7o2aqan7/open_tuner_0.b_20240709.zip?dl=0

PA0RWE, 2024-07-10:
Hi Ton,

 

I have tested the new version but still 125K DVB-S is not working.

66K and 250K are OK.

I have checked it with SDRAngel and there was a picture on 125K (see attached screenshot)

I also attached a screen shot from the status at 125K.

 

73’s

Rob PA0RWE

 

https://www.dropbox.com/s/2dcg7eq7o2aqan7/open_tuner_0.b_20240709.zip?dl=0

PA0RWE, 2024-07-31:
Hi Tom,

 

I have tested the 0B version from your site but found an error:

 

*	I don't get a picture on the first screen, sound is OK.

 

I have attached the relevant screenshots.

 

Besides,  the program stops after a while.

 

Version 0A is working great..!

 

73's

Rob PA0RWE

 

 

 

 

 

 
<https://www.dropbox.com/s/2dcg7eq7o2aqan7/open_tuner_0.b_20240709.zip?dl=0>
https://www.dropbox.com/s/2dcg7eq7o2aqan7/open_tuner_0.b_20240709.zip?dl=0

-
view it on GitHub, or
TECNAVCNFSM6AAAAABKUFHYSGVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDEMJZGYZDA
NBYHA> unsubscribe.

PA0RWE, 2024-08-01:
Hi Tom, I have tested version 0.b_20240709  but still no picture on 125K DVB-S
On 250 K upper and lower screens are OK. See attachments

[image]
[image]

PA0RWE, 2024-08-02:
I also found that the MER in the status is not OK (= 0.0)
[image]

PA0RWE, 2024-08-02:
Despite the different signals, the MER values ​​are the same. That doesn't seem OK to me. See screenshot
[image]

PA0RWE, 2024-10-09:
Hi Tom,

 

I have installed the latest version 

 

But still DVB-S in 125K is not working, no detection of the signal. I tested
my signal on SDRAngel and no problem.

No problem with other SR's.

I also asked a friend of mine to try OpenTuner on my signal with the same
result. No picture. Very strange isn't ?

 

I hope you can find the problem, and if you need more info, please ask me!

 

Best 73's

Rob PA0RWE

 

 

 

 

 

 

 

 

 

 

 

 

 

 

 
<https://www.dropbox.com/s/2dcg7eq7o2aqan7/open_tuner_0.b_20240709.zip?dl=0>
https://www.dropbox.com/s/2dcg7eq7o2aqan7/open_tuner_0.b_20240709.zip?dl=0

-

PA0RWE, 2024-11-08:
Hi Tom,

I have tested 2024/07/09 and 2024/07/14 with DVB-S @ 125K, but both are not working .
If you need more info, please ask me!
73's
Rob PA0RWE

PA0RWE, 2024-12-30:
Hi,
It seems that the problem is not OT!  A friend of mine have tested OT with 125K on Hailsat and no problem. He is using the FTDI setup while I'm using the PicoTuner. So I go back to the FTDI setup and test OT again on 125K. I will let you know the results.