tomvdb/open_tuner #42: Video su + monitor

Iw3gim, 2023-04-29:
Good morning,
congratulations for the software for receiving two channels of QO100, I wanted to propose some possible implementations, if possible.
1- In order to be able to see on a larger screen, it would be advisable to have the possibility of selecting the image from the two boxes present in the program and perhaps being able to move it to a second monitor connected to the PC.
2- the second image that remains on the porgramma also has the possibility of moving it to any other monitor or putting it on the internal monitor on the monitor where the program is open.
3- Possibility of being able to choose two video transmissions and not having the first channel always active on the beacon, once tuner 1 or 2 has been selected, it would be useful to be able to choose what to receive on that channel, by clicking on the signal present in the bar below.
I hope I explained myself, I don't know if it will be feasible to make these changes, but it seemed right to me to send you the idea given the great work you have done.

Hi Iw3gim
Alexander
ARI Feltre - www.arifeltre.it

tomvdb, 2023-04-29:
Hi Alexander,

Anybody can always send their ideas :) Thanks for that.

(1)(2) I had an idea of having the ability to "undock" the video from the app into a seperate window and then it should be possible to move it around. I'm going to have to experiment to see if I can do this while its playing video though. This might be a setting that would require a restart.

(3) I might be misunderstanding this one, you can already choose different signals for different tuners. The spectrum is divided into two, so you can select on the top side to choose 1 signal, and select another signal on the bottom side to choose another signal. You can also change both signal independently via the tuner window (press the small button with the T on in). Let me know if I miss understood this question.

Thanks for the feedback :)
73,
Tom

Iw3gim, 2023-04-30:
Hi Tom,
thanks for your answer.
Ok for the 1st point.
For the 2nd very good, I hadn't found the possibility of porter also changing the input of the 1st tuner.

Hello and congratulations again for the good work.
Alexander

tomvdb, 2023-04-30:
Please give new version a try 0.7c on website, I added some experimental window settings.

73,
Tom

closed 2023-04-30 by tomvdb