tomvdb/open_tuner #43: UDP multicast not compliant

ON1AVO, 2023-04-30:
Tom,

I've set the UDP outputs to a multicast address & different port(s) to verify the stream(s) with a real-time MPEG analyser but it doesn't sync.
VLC or PotPlayer seems to have no problem with the UDP stream.
Using the UDP output of MT or F1EJP FreeStreamCoder the analyser reports well.
Wireshark files, with UDP filter enabled, are available for analysing, atm too big to upload.

tomvdb, 2023-04-30:
Hi,

I can have a look, but the data is basically just passed along as received from the Minitiouner hardware. What are you using as your real time MPEG analyzer? I can setup on my side and investigate.

73,

ON1AVO, 2023-04-30:
Tom,

Software is StreamXpert from Dektec, linked to their own hardware.
TSReader(Lite) is perhaps an alternative.
Attached 3 screenshots:
1: StreamXpert build in NIC_OT: no sync through the build in NIC or an attached USB-NIC from OT
[image]
2: StreamXpert thru DTA2160_OT: sync through the dedicated DTA2160 IP input from OT
[image]
3: StreamXpert build in NIC_MT: sync through the build in NIC from MiniTioune
[image]
The UDP multicast from MiniTioune works fine without using the dedicated Dektec card.

73, Freddy ON1AVO

ON1AVO, 2023-05-01:
Tom,

 Wireshark screenshots, port 15000 used for Minitiouner and 10000 / 10001 for OT.
[image]
[image]

73, Freddy ON1AVO

tomvdb, 2023-05-01:
Thanks Freddy, Haven't had a chance to dig into this one yet, but its on my todo :)

ON1AVO, 2023-05-02:
Tom,

No hurry 😉
It is a feature not many OM's will use, I just use it to verify the timing
and errors in the transportstream(s).

73 Freddy
ON1AVO

G8YTZ, 2023-05-02:
Hi Tom,

I'm fining a similar issue. UDP using Minitioune software ok to VLC, using the same settings in OpenTuner and no reception. I can send a pcap file if you want, but I'm sure it will be similar to Freddy's.

Settings I'm using:

On Mintioune & OpenTuner
127.0.0.1
9080

On VLC:
udp://@:9080

Using H.264 transmission for the test

Objective of test was to move on to test using MPVNET player instead of FFMPEG for h.266 as suggested, but first testing a know working receiver.

Justin.

F5OEO, 2023-05-04:
Like tje original longmynd, the output is not mpeg compliant : 
- udp packets should be max 7 ts packet (7*188 bytes)
- udp packet should be sync aligned (0x47 at first byte)

See https://github.com/F5OEO/longmynd/commit/4b76824298bf76aaa8abeeacbe07d9f2ae3a030b for example

tomvdb, 2023-05-04:
> Like tje original longmynd, the output is not mpeg compliant :
> 
> * udp packets should be max 7 ts packet (7*188 bytes)
> * udp packet should be sync aligned (0x47 at first byte)
> 
> See [F5OEO/longmynd@4b76824](https://github.com/F5OEO/longmynd/commit/4b76824298bf76aaa8abeeacbe07d9f2ae3a030b) for example

Thanks Evariste, I'll have a look. I was wondering about the alignment as I had to do that to feed it to the mediaplayers in OT.

tomvdb, 2023-05-04:
@ON1AVO don't you want to give this test version a try please when you have a chance: https://www.dropbox.com/s/cx4tqsjlo2znqql/ot_test_20230504a.zip?dl=0

F5OEO, 2023-05-04:
Ok...this is working now ! Great !

[image]

ON1AVO, 2023-05-04:
Tom,

I can confirm like Evariste, the UDP multicast is now working with the internal NIC selected.

Tnx

73, Freddy