tomvdb/open_tuner #44: Local Oscillator offset not working

G8YTZ, 2023-04-30:
IF Offset that was fixed in an earlier release (01/04/23) e.g. Rx 3404000, LO 5150000 No reception

- Requested freq shows 741525 (IF tuned frequency should be 1746000)
- Symbol rate shows 1500 (even though 2000 is in the memory)

It might be that you need a tick-box to specify a +ve or -ve offset for the LO, similar to how the Ryde/Portsdown operates as 5150MHz LO is above the Rx frequency in this case.

Note: 929MHz offset is ok. So more-or-less confirms this is an issue where +ve

Basically I currently use 3 LOs:
9750MHz - QO100 (Not Tested by G8YTZ)
5150MHz - C-Band LNB for 9cm (Not OK)
929MHz  - IC-9700 23cm I st IF when using Radio Analog board PTRX9700 for 23cm reception (OK)

tomvdb, 2023-05-01:
Justin, is this while using the Tuner? or a stored frequency ?

G8YTZ, 2023-05-01:
Tom,

Yes, stored memory. Not tried with Tuner, a lot of those functions have gone on the 0.7 release.

Justin.

tomvdb, 2023-05-02:
Justin,

Can you send me the frequencies.json file please in the OT directory after you have configured those 3 offsets.

Thanks,
Tom

G8YTZ, 2023-05-02:
`[
  {
    "Name": "QO-100 Beacon Tom Test",
    "Frequency": 10491520,
    "Offset": 9750000,
    "SymbolRate": 1500,
    "RFInput": 2,
    "DefaultTuner": 0
  },
  {
    "Name": "up2409",
    "Frequency": 2409750,
    "Offset": 0,
    "SymbolRate": 333,
    "RFInput": 2,
    "DefaultTuner": 0
  },
  {
    "Name": "437 125 A",
    "Frequency": 437000,
    "Offset": 0,
    "SymbolRate": 125,
    "RFInput": 1,
    "DefaultTuner": 0
  },
  {
    "Name": "437 333 A",
    "Frequency": 437000,
    "Offset": 0,
    "SymbolRate": 333,
    "RFInput": 1,
    "DefaultTuner": 0
  },
  {
    "Name": "GB3JV - 5150 Tom Test",
    "Frequency": 3404000,
    "Offset": 5150000,
    "SymbolRate": 2000,
    "RFInput": 2,
    "DefaultTuner": 0
  },
  {
    "Name": "1249MHz - 929 - Tom Test",
    "Frequency": 1249000,
    "Offset": 929000,
    "SymbolRate": 1000,
    "RFInput": 1,
    "DefaultTuner": 0
  }
]`

tomvdb, 2023-05-04:
Hi Justin,

The offset parameters are for the physical offset required between the frequency you want to deal with and the end result that gets sent to nim tuner. Basically LO != offset, might look at making it smarter in the future but for now its just purely based on the offset directly.

73,
Tom

closed 2023-05-04 by tomvdb

G8YTZ, 2023-05-05:
The issue is thta you cannot specify the offset and negative or positive, ot a big issue though, I can always enter the IF frequency.