tomvdb/open_tuner #45: FFMPEG Player keeps crashing

G8YTZ, 2023-05-01:
Gareth G4XAT and I did some tests today with H.266 on 70cm and 2m at various symbol rates over 5.2km

- G4XAT=>G8YTZ G4XAT using DATV Easy v2.17 at various symbol rates between 125k  and 333k and various FECs
- G8YTZ using v0.7c with Minitiouner v2. PC is an Intel NUC8 i7-8559U 2.71GHz CPU, 32GB RAM 256GB SSD for OS, Win10 Pro 21H2 64 Bit

Problem is that FFMPEG player is very unstable, picture disappears or freezes after a few seconds, but TS record is ok. No issues with FFMPEG or VLC Player with H.265 

Note: Display device is 4k LG Monitor.

- It seems that any TS glitch upsets the player.
- I tried both embedded and external player.

I was thinking of trying mpv.net player seems to work well, perhaps that could be used?

tomvdb, 2023-05-01:
The h266 support is patched into ffmpeg, so its all still fairly glitchy and experimental. This newer version should work better, atleast on satellite, but not sure on terestrial. was there any specific error messages in the terminal ?

https://www.dropbox.com/s/lvbqzjlm9rvou68/open_tuner_0.7d_20230501.zip?dl=0

tomvdb, 2023-05-01:
I'm assuming you guys tried mpv.net as well via the udp port?

G8YTZ, 2023-05-01:
> I'm assuming you guys tried mpv.net as well via the udp port?

Tom, We did not get that far, but used it to replay the recorded ts files. I'll do some testing with that though.

Justin.

G8YTZ, 2023-05-01:
> The h266 support is patched into ffmpeg, so its all still fairly glitchy and experimental. This newer version should work better, atleast on satellite, but not sure on terestrial. was there any [...]

OK Tom, many thanks. I'll give it a go

Justin.

G8YTZ, 2023-05-02:
Tom,

this (0.7d) version looks much more stable. Only tested at 125k as that's the best my 6th Gen i7 HP laptop can do.

Justin.

tomvdb, 2023-05-04:
Closing this for now, feel free to create a new issue with any specific ffmpeg issues. ffmpeg is still fairly new integration and will be improved over the next couple of versions :)

73,
Tom

closed 2023-05-04 by tomvdb