tomvdb/open_tuner #46: Second Tuner shows allways the beacon when i switch to another station

DL2GHM, 2023-05-01:
Hi Tom,
the second Tuner in my case receiving the beacon. No Matter wich station i want to decode...
Any Idea?
Thank you for the fantastic progress!
It is a big step to have both tuners activated 💯 
Best regards
Martin
DL2GHM
[image]

tomvdb, 2023-05-01:
This is very odd, I don't see any issues and I can see that its tuning to the right signal. I did fix a small issue on the vlc side, not sure if its related, but please try this version: 

https://www.dropbox.com/s/lvbqzjlm9rvou68/open_tuner_0.7d_20230501.zip?dl=0

Looks like you are using the Minitiouner Pro 2 ?

Can you try to see what you receive over the udp port, or what it records when doing recording ?

tomvdb, 2023-05-01:
I'm assuming tuner 1 works fine, and if that is tuned to another signal then the beacon is still coming through perfectly on tuner 2 ?

DL2GHM, 2023-05-01:
Yes the upper Tuner works perfect...
The lower ignores the choice i make an shows the beacon.
The Tuner Data shows the D and other info from the choice i made...

g6hfs, 2023-05-01:
Tom,

Using my pro 2, I have no problems, both channels record and store the files in the opentuner folder, both have same date/time apart each files ends ts1 and ts2

Brian

DL2GHM, 2023-05-01:
Thank you Brian! Did you have to program the USB Ports or done anything?
I have done nothing till now. Using it out of the box...

g6hfs, 2023-05-01:
Nothing special, just plugged both pro 2 usb port B’s into laptop, run opentuner on different signals then press R and the Rec on each tuner started blinking.

Not tried batc v2 as waiting pcb to add second stream

Brian

DL2GHM, 2023-05-02:
Hi Brian,

tnx for your help. Thats the way i connected the MT pro 2 to my PC....

Martin

tomvdb, 2023-05-02:
Hi Martin,

ah, so you have two minitiouners plugged into the same pc? OT probably got confused and opened the TS of the other which was already receiving video. That makes total sense. I'm going to update OT so that you can specify serial numbers of devices that it needs to open, so you'll be able to prevent that from happening.

Thanks for letting me know, I was wondering about that all day :p

T

tomvdb, 2023-05-04:
Closing Issue as its resolved. Next version will have some additional functionality to specify specific usb devices based to better handle minitiouners on same pc.

73,
Tom

closed 2023-05-04 by tomvdb