tomvdb/open_tuner #50: Pi shut down button

ShaunG8VPG, 2023-05-06:
Tom,
Thank you for this wonderful software. On Longmynd Client, one useful feature would be a button to send a shut down command to the Pi. I have tried two scripts from online sources to operate a physical button, but cannot get either to work. An alternative might be if your software contained the script to operate a shutdown button between ground and a nominated GPIO pin.
73 Shaun G8VPG.

tomvdb, 2024-02-08:
Hi Shaun,

Apologies for the late answer. Yes, I can add that in. I'm planning a new release soon and will be sure to include that in.

73,
Tom

tomvdb, 2024-07-10:
Hi Shaun, 

I will be adding something in OT for this. Are you still using Longmynd client or have you moved over to OT. I'm looking for more testers on OT for lonmynd client usage.

T

ShaunG8VPG, 2024-07-10:
Tom,Thanks for replying. I am now mostly using Picotuner with OT.73 Shaun.