tomvdb/open_tuner #52: Remote Web access to tuning etc?

dogastus, 2023-05-08:
Would it be possible to add remote access to select a station or frequency? This is so that someone on the Internet could access a receiver remotely and use it with streaming.

tomvdb, 2024-02-08:
Yes, it is definitely possible to add this in. What did you have in mind? A websocket to send command from a webinterface or a complete web server builtin?

dogastus, 2024-02-10:
Just a method of setting freq and symbol rate. If there was a way of 'serving' the spectrum display over the web, a remote user could then select the desired station.

dogastus, 2024-02-14:
You mentioned Quicktune UDP integration in the latest release but I can't make it work. Any guidlines on what settings to use in Quicktune?

tomvdb, 2024-07-10:
> You mentioned Quicktune UDP integration in the latest release but I can't make it work. Any guidlines on what settings to use in Quicktune?

Sorry, I missed this question. Did you come right with this ? 

https://www.zr6tg.co.za/opentuner-quicktune-control/