tomvdb/open_tuner #57: Tuning / changing parameters

9H1LO, 2023-06-17:
It looks like the program is quite specific to QO100 use, i get it, and the functionality is great by the looks of it 

however, those not only running QO100, but use it for general DATV use I think it would be nice if we could easily "tune around" with the basic parameters like Freq, SR etc, the Tuner box doesn't seem to do anything for me and will only go down to 9750 it's like its assuming we always have an lnb lo offset

the only way i can manage to tune and set sr is with the Freq Manager, which is somewhat annoying on top of a hill in a field with the sun on the laptop screen, something like a simple transceiver style tuning display would be so nice, i am hopeless at windows development so can't contribute much, had a look at the C# sources though...makes some sense since i come from a C++ background but but visual studio and windows are greek to me :/

9H1LO, 2023-06-17:
ok i seem to have gotten the tuner box working, you need to always consider LO offset and press update, what about just editing these directly in the left panel ? also it seems to "auto everything"....SR, FEC, Mode etc, can they be "forced" manually instead of auto ? 

[image]

tomvdb, 2024-02-23:
Version 0.A has various improvements around the tuner and will be released soon :)

closed 2024-02-23 by tomvdb