tomvdb/open_tuner #61: ffmpeg update not working

F1FAQ, 2024-01-23:
Hello Tom,

I compiled ffmpeg (shared) and placed the result in the ffmpeg directory, but opening open_Tuner fails.
A windows opens for 1 second saying "loading ffmpeg libraries fails"
Any idea why my compilation (wide open / non free / "all codecs") is not working in your application, but
is working well on my PC ?
Or perhaps you can provide an updated ffmpeg ?
Many thanks for this software
73 de Christian / F1FAQ

tomvdb, 2024-02-08:
Hi Christian,

This is most likely due to version mismatch. Depending on the changes between versions my bindings will be different and then it will fail. What version of ffmpeg are you trying to update to ? I'm currently on:

`ffmpeg version n6.0-11-gda761f6cc7 Copyright (c) 2000-2023 the FFmpeg developers
  built with gcc 12.2.0 (Rev10, Built by MSYS2 project)
  configuration:  --disable-static --enable-shared --pkg-config=pkgconf --cc='ccache gcc' --cxx='ccache g++' --ld='ccache g++' --extra-cxxflags=-fpermissive --extra-cflags=-Wno-int-conversion --disable-autodetect --enable-amf --enable-bzlib --enable-cuda --enable-cuvid --enable-d3d11va --enable-dxva2 --enable-iconv --enable-lzma --enable-nvenc --enable-zlib --enable-sdl2 --enable-ffnvcodec --enable-nvdec --enable-cuda-llvm --enable-libmp3lame --enable-libopus --enable-libvorbis --enable-libvpx --enable-libx264 --enable-libx265 --enable-libdav1d --enable-libaom --disable-debug --enable-libfdk-aac --enable-libvvdec --enable-libvvenc --enable-openssl --extra-libs=-liconv --enable-gpl --enable-nonfree`

I can definitely look at upgrading it if better version is available.

T

tomvdb, 2024-02-09:
Cool, glad to hear. Any advantages to the newer ffmpeg dll files? Maybe I should include yours in the bundle (or stop being lazy and compile the latest).

Yeah, the current checked in version requires a bit more. I'll update the docs, etc. I'm busy adding support for Colin's picotuner (a work in progress replacement for the ftdi modules)

https://github.com/g4eml/PicoTuner/tree/main