tomvdb/open_tuner #62: Problems

Tk1cx, 2024-02-10:
Hello, 
my vrsion 0.8 open tunner crashed.
This is what it tells me.
System Culture Setting: fr-FR
Init done
False
Websocket: QO_Spectrum: Try connect..

Websocket: QO_Spectrum: Connected.

Restoring Window Positions:
 Size: (28,160)
 Position: (-32000,-32000)
Error reading frequencies.json - file missing, or wrong format :Le fichier 'C:\open_tuner_0.8_20230919\frequencies.json' est introuvable.
Load Done
Websocket Server Started on :9090

even after reinstalling ........
a solution?

many thank 73' Thierry

tomvdb, 2024-02-10:
Hi Thierry,

There was an issue with the websocket server after a windows update. Please try the latest test version: https://www.dropbox.com/s/wr0nhsbor1dsril/open_tuner_0.9_20240210.zip?dl=0

If you are still having the same problem then it might be something else.

T

Tk1cx, 2024-02-10:
Thank you problem solved

tomvdb, 2024-02-10:
Great :)

closed 2024-02-10 by tomvdb