tomvdb/open_tuner #63: Random crashing

dogastus, 2024-02-10:
You probably will need debug info but initially I'm reporting this as maybe there is a generic solution. When I run the app, it initially works fine but sometimes it will crash in a variety of ways. Sometimes, after successfully receiving a station, I click on another station and the program just hangs. Other times, it will randomly display the received video in a seperate window even though I haven't selected that option. Sometimes, it will crash completely, the whole app just exiting suddenly.

tomvdb, 2024-02-10:
Please try the latest test version: https://www.dropbox.com/s/wr0nhsbor1dsril/open_tuner_0.9_20240210.zip?dl=0

A few of the bugs you describe has been fixed, but not released yet as I'm still testing, but this should fix atleast the seperate window issue.

I'm going to need more debug info, preferably if it the issue can be replicated easily.

T

dogastus, 2024-02-14:
It's looking fairly stable right now.

tomvdb, 2024-02-15:
Cool, I'm closing this ticket for now, feel free to create a new one with specific issues :)

closed 2024-02-15 by tomvdb