tomvdb/open_tuner #65: Exception error with MPV set to open in separate window

dead-fets, 2024-02-11:
MPV as second tuner, box ticked to open in a separate wirndow. An exception error occurs and OT fails.

Had to load frequencies.json manually by copying from previous version's directory.

Great work Tom, good to see you returning to OT. Would it be simple to make the setting for UDP outputs sticky??

Thanks and all the best,, Chris 2E0ILY

tomvdb, 2024-02-12:
Hi Chris, 

Please try out new test version with various fixes as reported by your 3 tickets. 

[Test Version](https://www.dropbox.com/scl/fi/c1anvv7pi11qv12po6cm4/open_tuner_0.9test_20240212.zip?rlkey=xyinrweo5fxckgz47kp9h1eax&dl=0
)
UDP outputs? you mean sticky on the enabled flag, or sticky on its settings ?

dead-fets, 2024-02-12:
Hello tomvdb/open_tuner,

 Monday, February 12, 2024

 Will try the update later, thanks Tom!
 
I meant it would be helpful if the UDP output settings for tuners one and two could be made 
to hold their enabled or disabled settings after closure and re-opening of OT.

Best regards,
 Chris Wilson   ***@***.***

> Hi Chris,
> Please try out new test version with various fixes as reported by your 3 tickets.
> Test Version
> UDP outputs? you mean sticky on the enabled flag, or sticky on its settings ?
------------0471030850C429915
Content-Type: text/html; charset=utf-8
Content-Transfer-Encoding: 8bit

<html><head>            <style type="text/css" title="rt_noDelete">
    blockquote.rt {
    margin: 0 0 15px;
    border-left: 4px solid #81c784;
    padding: 0 0 0 12px;
    display: block;
    }
    p { margin: 0 0 0 0 }
.email-signature {font-family:"Arial"; font-size: 12pt; font-style: italic; font-weight: normal; text-decoration: none; }
    </style></head><body><p class="norm"><br/></p><p class="norm">Hello tomvdb/open_tuner,</p><p class="norm"><br/></p><p class="norm"> Monday, February 12, 2024</p><p class="norm"><br/></p><p class="norm"> Will try the update later, thanks Tom!</p><p> </p><p>I meant it would be helpful if the UDP output settings for tuners one and two could be made </p><p>to hold their enabled or disabled settings after closure and re-opening of OT.</p><p class="norm"><br/></p><p class="norm"><br/></p><p class="norm">Best regards,</p><p class="norm"> Chris Wilson    <a class="HR" ***@***.******@***.***</a></p><p class="norm"><br/></p><p class="norm"><br/></p><p class="norm"></p><blockquote class="rt"><p></p>
<p dir="auto">Hi Chris,</p>
<p dir="auto">Please try out new test version with various fixes as reported by your 3 tickets.</p>
<p dir="auto"><a href="https://www.dropbox.com/scl/fi/c1anvv7pi11qv12po6cm4/open_tuner_0.9test_20240212.zip?rlkey=xyinrweo5fxckgz47kp9h1eax&amp;dl=0" rel="nofollow">Test Version</a><br/>
UDP outputs? you mean sticky on the enabled flag, or sticky on its settings ?</p>

<p style="font-size:small;-webkit-text-size-adjust:none;color:#666;">—<br/>
<script type="application/ld+json">[
{
"target": "https://github.com/tomvdb/open_tuner/issues/65#issuecomment-1938511963",
"url": "https://github.com/tomvdb/open_tuner/issues/65#issuecomment-1938511963",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]</script></blockquote></body>
------------0471030850C429915--

dead-fets, 2024-02-12:
Fixed in test version, excellent and very fast work Tom, thank you.