tomvdb/open_tuner #70: open_tuner_0.9_20240211

G3GJA, 2024-02-19:
RF input level indication is stuck permanently at -82dBm.

Switching between A and B sockets makes no difference to the displayed amplitude, but the MER changes.

If, however the same switching is done using the Minitioune software the displayed amplitude changes between -20dBm and -67dBm.

Thanks for your work on this project.

Clive, G3GJA

tomvdb, 2024-02-23:
Hi Clive,

Thanks for reporting. I did some tests and you are correct. I'm working on a fix.

T

dl1rf, 2024-12-07:
> RF input level indication is stuck permanently at -82dBm.
>
> Switching between A and B sockets makes no difference to the displayed amplitude, but the MER changes.
>
> If, however the same switching is [...]

Hi Brian,

I have fixed  "RF level stuck at -82 dB" in source code.
Fix shall be include in next release.

73 de Martin, DL1RF