tomvdb/open_tuner #71: display problem

Tk1cx, 2024-02-24:
Good morning, 
 I often receive good signals at the spectrum but I can't get the video to display.
Is there an explanation?
Thanks Thierry

tomvdb, 2024-02-26:
Hi Thierry,

I'm going to need more information please. What hardware are you using? Are you looking at signals on QO-100 or terrestrial? What is the Mer and db Margin on the signals? Remember the spectrum is not from your device, but rather from the internet via the batc spectrum (https://eshail.batc.org.uk/wb/)

If you share some more info then I can help you more :)

Thanks,
Tom

Tk1cx, 2024-02-28:
Hello,
minitiouner for receiver
parbole1.2m
LNB 18v
reception on QO-100 
sea all depends on sations
Db same all depends on stations
yes I know that the spectrum comes from the internet
voila TOM

and thank you for your reply

tomvdb, 2024-04-08:
Apologies, I missed your reply. Did you come right with this ? If not, can you send some screenshots where its locked on the signal, but no picture.

T

tomvdb, 2024-04-15:
I'm closing ticket for now, but if you still haven't come right then feel free to open a new ticket with all the info.

73,
Tom

closed 2024-04-15 by tomvdb