tomvdb/open_tuner #77: Feature request: AUTOMUTE function on startup

DL2GHM, 2024-04-10:
Would be nice to AUTOMUTE all channels when starting up. CW Beacon Volume nearly kills the speakers sometimes after restarting.
Best regards DL2GHM

dl1rf, 2024-04-14:
I am working on this issue within my fork.
Goal is to save the mute and premute states within the settings.

DL2GHM, 2024-04-14:
Thank you for the info…

tomvdb, 2024-04-15:
Thanks Martin, PR has been merged. I'm fixing a few other things and will then do another release in the next couple of days.

T

DL2GHM, 2024-04-16:
> The feature implementation is finished.
> The following functionality will be provided:
> Mute Button will go to red if muted.
> Mute state and volume state before mute will be saved on program exit.
> A new check box at settings page will force mute at startup if checked.
>
> vy 73 de Martin, DL1RF

Thank you very much!
Especially for the last feature. It is important because it is also working after restarting after a crash.
73 de DL2GHM
Maritn

tomvdb, 2024-07-09:
Closing Issue, this functionality is now all in 0.B thanks to Martin (DL1RF) - Feel free to create a new issue if I missed something.

closed 2024-07-09 by tomvdb

DL2GHM, 2024-07-09:
👍