tomvdb/open_tuner #79: Opentuner 0-A starts then crashes, log file linked

dead-fets, 2024-04-12:
OT 0-A worked fine for about two days, now it launches and upon pressing connect button crashes. Log file is available at http://www.chriswilson.tv/log.zip

I tried deleting this version and re-installing. I *THINK* the trouble starts when changing the order of decoders from the default, or changing the saved recordings / snapshots folder.

Thanks Tom.

DL2GHM, 2024-04-12:
Same issue in my three installations of OT 0A. Setting up to ffmpeg results in crash after 2 seconds.

tomvdb, 2024-04-15:
Thanks, I'll have a look at this today. It definitely helps to have a reproducible problem. Will have feedback tonight

tomvdb, 2024-07-10:
Closing this as all should be good with the new 0.B version, let me know if these crashes still happen

closed 2024-07-10 by tomvdb