tomvdb/open_tuner #85: Request

dead-fets, 2024-04-19:
Not sure if this is the appropriate place for requests, but here goes.

PLEASE put the signal report buttons back on the main window and not in the chat Tom! 

Having the two windows open on a laptop isn't always a good setup IMHO, the old way, albeit a button for each tuner when using a two RX Mintiouner or whatever was much less fiddly ;) 

Thanks for looking, any supporters of this reversion??.

F1FAQ, 2024-04-22:
Tom,
Could we get back the option of V0.9 : Frequency manager ?
Many om's complain of this missing functionality 
Thanks / Christian

F1FAQ, 2024-04-24:
Tom,
I have two (and more) minitiouners. So I would like to be able to open OT twice to get up to four "video channels" udp outputs.
Could this be implemented ? (yes I know Winterhill exist!) 
Thanks / Christian

dead-fets, 2024-04-28:
When recording the various nets the lack of the old "Record All" option is apparent in the current release, and a PITA :) Will you perhaps be able to add it again please Tom? Thanks.