tomvdb/open_tuner #92: Tuning after a Winterhill reboot

m1dst, 2024-06-19:
I notice that after a Winterhill reboot, I am unable to tune to any stations even with the QuickTune feature enabled.  If I load the standalone QuickTune (v1.3) app and click on a station, then Open Tuner starts decoding fine.  Any future tuning requests can now come directly from Open Tuner from that point on.

There are a number of options available in QT that aren't in OT and I wonder if in my case this is the issue.  In QT I am able to enable the 18V voltage.  In OT there is no such setting.  My guess is that once it is enabled in QT, OT doesn't change it and therefore it works.

Also sending it the current IP address rather than having to right click on TS Address and setting it would also be super useful and save time after a reboot.

tomvdb, 2024-07-10:
Hi James,

Thanks for the feedback. I'm looking into these items and should have a test version soon.

T

dl1rf, 2024-12-06:
I see the same problem.
I will address it in the next days and will try to fix it.

vy 73 de Martin, DL1RF