tomvdb/open_tuner #93: Frequent Crashes, VLC issues? (logs attached)

MalinkyZubr, 2024-07-01:
[ot_log_2024-08-4--08-16-45.txt](https://github.com/user-attachments/files/16058594/ot_log_2024-08-4--08-16-45.txt)
[ot_log_2024-03-3--14-44-35.txt](https://github.com/user-attachments/files/16058595/ot_log_2024-03-3--14-44-35.txt)
[ot_log_2024-01-7--12-53-14.txt](https://github.com/user-attachments/files/16058596/ot_log_2024-01-7--12-53-14.txt)

After a few days of use with OpenTuner 0.A 2024/04/09, I ran into frequent crashing issues. Crashes only occurred when locked and displaying an image. After disabling hardware decoding and some other VLC settings as suggested by [this](https://forum.videolan.org/viewtopic.php?t=156428) guide, the problems went away. However a week later, the problems returned even after a fresh install of both VLC and opentuner. Was wondering if there were any known solutions to this.

Thanks, other than that I love opentuner! Very good and simple to use

MalinkyZubr, 2024-07-02:
As an update, I tried running opentuner with a minitioune unit and ran into the same issues. Minitiouner works with the minitiouner software as well.

tomvdb, 2024-07-08:
Hi,

Thanks for the logs, please try with this latest version and see if there is improvement.

Is this VLC specific, ie. if you are using mpv is it better?

tomvdb, 2024-07-10:
Sorry, forgot to post the link: https://www.dropbox.com/s/2dcg7eq7o2aqan7/open_tuner_0.b_20240709.zip?dl=0

MalinkyZubr, 2024-07-15:
Hi,
I only just got the chance to test the new version today. Minitiouner with OpenTuner works perfectly on a Windows 11 laptop and Windows 10 desktop. VLC doesn't seem to be putting up a fuss anymore. Aside from those two, I have a Windows 10 machine that isn't able to open the program and doesn't generate any logs. Not entirely sure what would be causing that. Are there any dependencies I should try manually installing?

When I get the opportunity to test with PicoTuner, I will update

Other than that, I was very glad to see the new preset feature. I was hoping for something like that. Very useful

MalinkyZubr, 2024-07-29:
Works perfectly. Thanks for your assistance!

closed 2024-07-29 by MalinkyZubr