tomvdb/open_tuner #94: Opentuner 0.B 20240707

G8YTZ, 2024-07-07:
Tom

Cannot start transport stream as the TS Address is fixed to 192.168.0.176 - software does not recognize the actual picotuner ip address (in my case 192.168.0.221)

Justin.

tomvdb, 2024-07-08:
Hi Justin, did you come right? The description of this ticket is currently empty ?

T

G8YTZ, 2024-07-08:
Yes,

I had an issue, but it turned out to be ports. I wanted to delete the comment, but could not.

The software is working well, very stable on Ethernet too.

Best regards,

Justin Cockett
***@***.***
www.gb3jv.co.uk <http://www.gb3jv.co.uk/>

tomvdb, 2024-07-08:
Glad to hear, I'll close the ticket. Thanks Justin.

T

closed 2024-07-08 by tomvdb

G8YTZ, 2024-07-08:
Sent from my Remington Rand UNIVACwww.g8ytz.com