tomvdb/open_tuner #96: Test vrersion B crashes on attempting to start with exception error

dead-fets, 2024-07-09:
Fresh download of test version B on Windows 10 Professional. 

Exception error on trying to start the app with the Connect button. Zipped copy of error here:

http://www.chriswilson.tv/error.zip

Ran as Administrator and as a normal user, same thing.

Thanks Tom.

tomvdb, 2024-07-09:
Hi Chris,

The error seems to indicate that the settings didn't load properly.

What hardware are you using? Did you replace the settings files with older ones or using the ones that came with the zipped folder?

Did it generate a log file in the logs folder ? if so can you send me that please.

73,
Tom

dead-fets, 2024-07-09:
It's a fresh install in its own directory Tom. I ran it again, after clearing old logs and this is what it logged after another exception error.

So I believe it's using the settings in the freshly downloaded zip of version B?

The PC runs a AMD Ryzen 9 5950X and the video card is an NVIDIA GTX 1070 

All previous versions have run without this error, if that helps at all?

http://www.chriswilson.tv/ot_log_2024-09-7--10-57-36.zip

Thanks Tom.

tomvdb, 2024-07-09:
Interesting, the log shows that its not reading two of your setting files properly. 

(1) Can you confirm they are both there? (minitiouner_settings.json and longmynd_settings.json)
(2) Can you have a look inside them and see if anything looks funny or corrupted?
(3) The files should actually generate if not found, so it would be weird if (1) is not there. Lastly can you delete those two files if they are there and run again so that it generates those files again.

[image]

I can't reproduce it on my side.

>The PC runs a AMD Ryzen 9 5950X and the video card is an NVIDIA GTX 1070

Sorry, I meant which minitiouner hardware

dead-fets, 2024-07-09:
Ahh, I just run the BATC Minitiouner with two FTDI modules fitted. I saw that OT was set by default to use Minitiouner Variant so left it at that.

Will check the files and report back Tom, thanks.

dead-fets, 2024-07-09:
Both of those files are in the settings directory Tom. They open with Visual Studio, one has nothing in it, the other the below. I'll delete both and see if they regenerate and things change!

  "media_path": "C:\\satmedia\\",
  "enable_spectrum_checkbox": true,
  "enable_chatform_checkbox": true,
  "enable_mqtt_checkbox": false,
  "enable_quicktune_checkbox": false,
  "enable_pluto_checkbox": false,
  "default_source": 0,
  "mute_at_startup": true,
  "auto_connect": false,
  "mediaplayer_preferences": [
    0,
    2,
    2,
    2
  ],
  "mediaplayer_windowed": [
    false,
    false,
    false,
    false
  ],
  "streamer_udp_hosts": [
    "127.0.0.1",
    "127.0.0.1",
    "127.0.0.1",
    "127.0.0.1"
  ],
  "streamer_udp_ports": [
    5000,
    5001,
    5002,
    5003
  ],
  "gui_window_width": 1490,
  "gui_window_height": 1156,
  "gui_window_x": 328,
  "gui_window_y": 118,
  "gui_main_splitter_position": 436
}

tomvdb, 2024-07-09:
Hi Chris,

Definitely sounds like something went wrong with your download. Those files aren't empty in the zip file. Either way it allowed me to add another check where if the file is corrupted then it will use the defaults for the settings.

New version to download and try:

https://www.dropbox.com/s/2dcg7eq7o2aqan7/open_tuner_0.b_20240709.zip?dl=0

I've also added some stuff inspired by George's fork.

- can hide/show the properties panel using CTRL-P shortcut
- double click on video to show overlay of basic info
- use mousewheel to change volume (with visual indication)

[image]

T

dead-fets, 2024-07-09:
The vesrion you linked to last, above works fine thanks Tom. I guess something occurred either in the downloading or the unzipping.

Is it to be expected that using the second tuner under FFMPEG the beacon doesn't decode, and I believe *NO* H264 sigs will decode using FFMPEG? That's been the case for 0.9 and maybe earlier, too. MPV and VLC seem to decode H264 / the beacon with no issues.

Here's a screenshot showing MPV decoding and showing the beacon video on tuner 1, but not ffmpeg on tuner 2

http://www.chriswilson.tv/no-beacon.jpg

Thanks for your patience and the new releases Tom!! :)

tomvdb, 2024-07-10:
Great, yeah something def went wrong somewhere. All good.

Yes, the ffmpeg and beacon is still an issue I'm looking at. Its not just you, its a global thing. I have a few ideas around that, but working through the list :)

I'm closing this issue for now, but if you find other issues feel free to open a new one.

73,

closed 2024-07-10 by tomvdb