tomvdb/open_tuner #98: Chat signal report shows DD instead of just D

dead-fets, 2024-07-11:
I am pretty sure it never used to do that? I recall the sig report just showed D*

Now it shows DD.  I'm sure there's a joke in there somewhere....   ;)

[image]

tomvdb, 2024-07-11:
there is some inconsistencies between the different sources and how they pass the data. what hardware are you using in this example ?

tomvdb, 2024-07-11:
Chris, if you have my email, can you send me a direct email please. Have something else I want to test :)

T

dead-fets, 2024-07-11:
I am using Minitiouner with two FTDI modules.

I don't have your direct mail Tom, but if you fire *me* a mail at chris at chriswilson.tv I will have? Cheers.

tomvdb, 2024-07-12:
This should be sorted :)

73,
Tom

closed 2024-07-12 by tomvdb