tomvdb/open_tuner #105: OT: Command line parameter to start AutoTune

EB2CTA, 2024-10-18:
Hi Tom:

At Radio Club Foronda (EA2RCF), we are promoting QO100 DATV. The project aims to showcase live DATV channel scans 24/7 on [www.ea2rcf/QO100/DATV](url) for radio amateurs and the general public interested in amateur TV. It also acts as a reception monitor for radio amateurs broadcasting DATV.

We are encountering a problem where the program stops after receiving for an extended period. We suspect a buffer overload causing it to crash. To address this, we stop the program every 30 minutes and restart it using the Windows task scheduler.

However, the program starts on the beacon and requires manual intervention to begin scanning. We kindly request the following adjustments:
1) AutoTune (Timed) program startup
2) Option to adjust scan delay time between channels (in seconds)
3) Default beacon selection in the absence of an amateur signal

Configuring these parameters in opentuner.exe command line or a .ini file would be greatly appreciated.

Thank you,
73 de Lino, EB2CTA | KO6CMV

dl1rf, 2024-12-06:
Hi Lino,

if Tom accepts my new code the requested function will be included in the next release.
The AutoTune (Timed) function will show the beacon if no other signal is present.
With other signal present it is possible to skip the beacon.
The settings for timing etc. can be set up within an new setup menu.
A command line option is not yet provided for the function but may be included if required.
Just drop a note here if you need a command line option.

73 de Martin, DL1RF

EB2CTA, 2025-02-10:
Hi Martin, We have made some modifications to Tom's code and we already have the ADTV automatic channel scanner up and running at [https://www.ea2rcf.org/QO100/DATV/](https://www.ea2rcf.org/QO100/DATV/)
We would like to try your new code as well.
73 de Lino, EB2CTA | KO6CMV

dl1rf, 2025-02-13:
Hi Lino, nice to hear that you get your ADTV automatic channel scanner up and running.
If you wish to compile and try my latest OT code please check my "devel" branch at [https://github.com/dl1rf/open_tuner/tree/devel](https://github.com/dl1rf/open_tuner/tree/devel).
73 de Martin, DL1RF

EB2CTA, 2025-02-14:
Thanks Martin, we will study your code carefully. :-)