tomvdb/open_tuner #108: ErrorWebSocketSharp

immusallasta, 2024-12-03:
Hi, 

I'm using OT 0.B with Minitiouner Pro V2, and is my problem:

Time to time error in Open Tuner v.0.B terminal, and after this OT freezes.
... and ones occured, this appears repeatedly.
Have to close terminal X in the right upper corner, restart.

73 de Ismo, oh1nhw

[10:29:25 INF] Websocket: QO_Spectrum: ErrorWebSocketSharp.ErrorEventArgs
[10:29:25 INF] Websocket: QO_Spectrum: ErrorWebSocketSharp.ErrorEventArgs
11.11.2024 10.29.25|Error|WebSocket.messagec|System.InvalidOperationException: The object is in use elsewhere. 
 at System.Drawing.Graphics.Clear(Color color)
 in opentuner.ExtraFeatures.BATCSpectrum.BATCSpectrum.drawspectrum(UInt16[] fft_data) in C:\source\tuner_tests\open_tuner\ExtraFeatures\BATCSpectrum\BATCSpectrum.cs on line 379
 at WebSocketSharp.Ext.Emit[TEventArgs](EventHandler`1 eventHandler, Object sender, TEventArgs e)
 at WebSocketSharp.WebSocket.messagec(MessageEventArgs e)
11.11.2024 10.29.25|Error|WebSocket.messagec|System.InvalidOperationException: The object is in use elsewhere.
 at System.Drawing.Graphics.Clear(Color color)
 in opentuner.ExtraFeatures.BATCSpectrum.BATCSpectrum.drawspectrum(UInt16[] fft_data) in C:\source\tuner_tests\open_tuner\ExtraFeatures\BATCSpectrum\BATCSpectrum.cs on line 379
 at WebSocketSharp.Ext.Emit[TEventArgs](EventHandler`1 eventHandler, Object sender, TEventArgs e)
 at WebSocketSharp.WebSocket.messagec(MessageEventArgs e)

dl1rf, 2024-12-06:
Hi,
I saw the problem too.
I debugged the issue and changed the drawing function.
The issue should be fixed with the next release.
With my test version I do not see the issue any more.

73 de Martin

immusallasta, 2024-12-06:
Hi,

Internet should be fine, but how to check if there is short interruptions?
I'll post later details from the terminal....

ismo

immusallasta, 2024-12-06:
Here is what's in the terminal of the reporter issue:

[16:26:45 WRN] DATV Reporter Disconnected
[16:26:55 WRN] DATV Reporter : Attempt Reconnect
OH1NHW 6th Dec -24

Datv reporter connect, disconnect, reconnect....

datv-spotter: Connecting: wss://datvreporter.info:3000
[16:27:05 WRN] DATV Reporter : Attempt Reconnect
datv-spotter: Connecting: wss://datvreporter.info:3000
[16:27:15 WRN] DATV Reporter : Attempt Reconnect
datv-spotter: Connecting: wss://datvreporter.info:3000
datv-spotter: Connected
datv-spotter: Connected
datv-spotter: Connected
[16:27:16 INF] {
  "target_callsign": "A71A",
  "observer_callsign": "OH1NHW",
  "observer_gridlocator": "KP00uv",
  "observed_time": "2024-12-06T14.27.16Z",
  "application": "OpenTuner",
  "application_version": "0.B - 2024/07/14 11:42:03.40",
  "db_margin": 3.7,
  "mer": 8.4,
  "frequency": 10491525,
  "symbol_rate": 1500,
  "hardware": "Minitiouner Pro 2"
}
6.12.2024 16.27.16|Fatal|<>c__DisplayClass9.<ReadBytesAsync>b__8|WebSocketSharp.WebSocketException: The header of a frame cannot be read from the stream.
                            kohteessa WebSocketSharp.WebSocketFrame.processHeader(Byte[] header)
                            kohteessa WebSocketSharp.WebSocketFrame.<>c__DisplayClassa.<readHeaderAsync>b__9(Byte[] bytes)
                            kohteessa WebSocketSharp.Ext.<>c__DisplayClass9.<ReadBytesAsync>b__8(IAsyncResult ar)
[16:27:16 WRN] DATV Reporter Disconnected

dl1rf, 2024-12-07:
Hi ismo,

ok, that look strange.
I will look deeper to into the problem.
As told with my last version I do not see that problem.
I will repeat testing wit original 0.B version.

73 de Martin, DL1RF